gazebosim/gz-sensors #17: Audio sensor and source

chapulina, 2020-04-28:
Implement multiple sound sources and microphones. Take a look at OpenAL, and see if there are other libraries that we could use.

adlarkin, 2020-09-30:
I spoke with @nkoenig about how we want to go about implementing this. I'm trying to start with something trivial like the [LogicalCameraSensor](https://github.com/ignitionrobotics/ign-sensors/blob/master/include/ignition/sensors/LogicalCameraSensor.hh). I looked at OpenAL, and although it could be used to produce actual audio to something like computer speakers or headphones, there doesn't seem to be a way to programmatically determine whether or not a listener/microphone can hear sound emitted from a particular source or not (if a sound device can hear the audio, OpenAL will output sound, but if not, then you just won't hear anything emitting from your sound device).

So, in order to programmatically determine whether a microphone can actually hear an audio source at a certain location or not, I'm thinking I could go down the following route:
* Define a graph of audio objects, with an audio object being either a "microphone" or source"
* In this graph, microphones can be connected to multiple sources, and sources can be connected to multiple microphones, but microphones cannot not be connected to other microphones, and sources cannot be connected to other sources
    - The cost of an edge between a source and microphone could be the euclidean distance of a straight line between the two objects (this wouldn't account for things like obstacles in the world, but we can find ways to create a better edge cost later)
* A microphone considers sound emitted from a source as "detected" if the sound pressure at the microphone is above a certain threshold
    - If we know the sound pressure emitted from the source (`p_1`) at a specific distance (`r_1`), then we can calculate the sound pressure (`p_2`) at the microphone's distance from the source (`r_2`) using `p_2 = ((r_1 / r_2) * p_1)`

@nkoenig @chapulina what do you guys think of the approach I described above? This won't emit actual audio to speakers or headphones, but I'm assuming that we are placing higher priority on a sensor that can mimic sound detection (plus, playing audio to an audio device from multiple sources at different locations could be tricky, and doesn't seem to be something we are trying to accomplish in simulation). My only concern is how we would define `p_1` and `r_1` for a particular audio source - perhaps those are parameters that the user needs to specify when creating new audio sources? I'm not aware of any example audio sources with these values that we could use for testing.

nkoenig, 2020-10-02:
I'm not sure why a graph is required. The microphone and source both have a pose in the world. You can compute the Euclidean distance between a microphone and all sources. Using some heuristic, the microphone would receive a signal. Is the graph serving a different purpose?

adlarkin, 2020-10-02:
> I'm not sure why a graph is required. The microphone and source both have a pose in the world. You can compute the Euclidean distance between a microphone and all sources. Using some heuristic, the microphone would receive a signal. Is the graph serving a different purpose?

Thanks for pointing that out - I don't think we need a graph since a microphone and source each have a pose, like you said. A lingering concern I still have is how we will compute whether a microphone can detect audio coming from a source or not - we could use the pressure formula that I mentioned in my [previous comment](https://github.com/ignitionrobotics/ign-sensors/issues/17#issuecomment-701619558) (based on the reading I did, it seems like most real-world microphones use pressure), but I'm not sure how we would come up with `p_1` and `r_1`. Does anyone know how we could get these values? If not, we may need to use something other than pressure.

chapulina, 2020-10-02:
I'm not up-to-date with what you are trying to accomplish, what are the requirements for your audio sensors?

Have you looked into the OpenAL source and sink [implementations](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/util/OpenAL.hh) on Gazebo classic? 

Users can attach [audio sink or sources](http://sdformat.org/spec?ver=1.7&elem=link#link_audio_sink) to links on SDF, as the link moves it [updates the sink and source positions and velocities](https://github.com/osrf/gazebo/blob/10863ffa8ec28e8e0e3d3a3a588d21cdc78055ab/gazebo/physics/Link.cc#L589-L604), then OpenAL APIs are used to propagate the sounds from sources to sinks based on various parameters.

adlarkin, 2020-10-02:
> I'm not up-to-date with what you are trying to accomplish, what are the requirements for your audio sensors?

@nkoenig feel free to correct me if I'm incorrect, but from my understanding, I want to create a sensor that can detect when audio has been received from a sound source (which is basically a microphone). I don't believe that a requirement is to actually play sound from a source in simulation to an audio device like speakers or headphones.

> Have you looked into the OpenAL source and sink implementations on Gazebo classic?
>
> Users can attach audio sink or sources to links on SDF, as the link moves it updates the sink and source positions [...]

Yes, I did take a look at the OpenAL source/sink implementation in Gazebo classic. I don't believe that this is what we want to achieve with audio sensors in ignition for two reasons:
* The source/sink implementation in Gazebo classic only supports one source (again, @nkoenig if this is incorrect, please correct me here), and we want to support multiple sinks and sources with this new implementation.
* OpenAL is great for rendering actual sound to an audio device (speakers/headphones), but there's no way to use OpenAL to see if the listener actually received sound from the source(s) or not - if the listener receives sound from a source, audio is played to a device, but if the listener does not receive sound from a source, then no audio is played to a device - we need a way to check in code whether a microphone has received/detected sound or not, but OpenAL is meant more for applications like games where audio only needs to be rendered to an audio device.

@chapulina I hope this clears things up, and hope that it's easier to understand my proposals in my other comments now. Let me know if you have any other questions.

chapulina, 2020-10-02:
Thanks, @adlarkin , this is clearer. It does sound very similar to the logical camera.

I'd say that the sensor you come up with wouldn't close this issue though. It would still be interesting to support actual audio for applications like speech recognition.

> programmatically determine whether a microphone can actually hear an audio source at a certain location

I'm not sure how much realism you need from this part. Video game engines usually offer customizable [sound attenuation functions and shapes](https://docs.unrealengine.com/en-US/Engine/Audio/DistanceModelAttenuation/index.html).

adlarkin, 2020-10-05:
> I'd say that the sensor you come up with wouldn't close this issue though. It would still be interesting to support actual audio for applications like speech recognition.

That makes sense :+1: 

> I'm not sure how much realism you need from this part. Video game engines usually offer customizable sound attenuation functions and shapes.

Thanks for sharing that link. So, if I understand correctly, @chapulina you're suggesting that I could do the following: 
* Make a `LogicalAudioSensor::Source` object that has the following attributes (perhaps we can add more later):
    - _Attenuation Function_
    - _Attenuation Shape_
    - _Inner Radius_
    - _Falloff Distance_
    - _Initial Volume Level_ (which would be somewhere between 0-100)
        - This would allow for different audio sources to play at different volumes (for example, perhaps one audio source plays audio that is a lot quieter than audio that is emitted from another source)
* Make a `LogicalAudioSensor::Microphone` object, which would probably just have the following attribute:
    - _Volume Detection Threshold_ (which would be somewhere between 0-100)
        - This just defines how much of an audio source needs to be heard by a microphone in order to count it as "detected audio"

If you agree with how I defined the `LogicalAudioSensor::Source` and `LogicalAudioSensor::Microphone` objects above, then I'm assuming that these attributes could be set in the world's `sdf` file. Feel free to let me know if I'm missing something, or if there's a better way to set the attributes for these objects.

chapulina, 2020-10-05:
Thanks, @adlarkin , that looks reasonable to me. I'd just recommend implementing only one attenuation method to begin with, but maybe leaving room to add more later _if needed_. I'd go with whatever function + shape can be physically justified and expose the relevant parameters through SDF. If you implement this as a plugin, as opposed to a sensor, you can add arbitrary parameters to it without making changes to the SDF spec, like the optical tactile sensor is doing in https://github.com/ignitionrobotics/ign-gazebo/pull/229.

adlarkin, 2020-10-05:
Sounds good; I'll go ahead and remove myself as the assignee for this issue since this `LogicalAudioSensor` doesn't seem to solve the goal of actual audio. We can move further discussion about the `LogicalAudioSensor` to the relevant PR once I open it. Perhaps this issue can be re-visited by myself or someone else at a later time - ideally, it would be nice to have some components from the `LogicalAudioSensor` be re-usable for actual audio sensors :grinning: