gazebosim/gz-sensors #19: Sonar

chapulina, 2020-04-30:
Port Gazebo-classic's [SonarSensor](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/sensors/SonarSensor.cc) to Ignition.

It should be integrated with the noise system.

Once this is done, a visualization can be added to `ign-rendering`.

felixstaniek, 2021-02-28:
Hi, I am currently trying to set up a simulation environment for autonomous model cars in ignition.
For this ultrasonic range sensors are required.
Since I do not have a complete overview over all elements available in citadel, i might be missing it, but has this been worked on and is there a branch where a sonar sensor can be used in ignition?

iche033, 2021-03-02:
no unfortunately, no work has been done on this yet.

mabelzhang, 2021-05-29:
Additional features for sonars --

I debated whether to put this in the RaySensor close-the-gap ticket (https://github.com/ignitionrobotics/ign-sensors/issues/26) or this one. I decided this one, because this ticket is for a sensor called sonar, while the RaySensor could be used for other things, even though users actually adapt the RaySensor to model sonars.

It has been raised that the ray / sonar sensors in Gazebo only return distances, while real sonar data depend on many other properties. To improve the sensor in Ignition, it would be great to support custom properties such as
- roughness
- material (not as in visual texture, but as in metal vs not metal, etc.)
- physical light scattering
and other things someone who actually uses sonars would be much more well-versed in than me.

We don't necessarily have to implement all of those (though that would be the ideal), but the idea is to have some kind of built-in support to make specifying a few properties easier, without users having to each write their own plugins before having a simulated sensor that meets common expectations.

Ryanf55, 2025-05-15:
Would this apply to a standard unidirectional rangefinder sensors commonly used on drones?
https://ardupilot.org/plane/docs/common-rangefinder-landingpage.html#unidirectional-rangefinders

B-grande, 2025-06-10:
Is this still in development? Or has this just dropped off the radar for now?