gazebosim/gz-sensors #24: Wide angle camera (fisheye)

chapulina, 2020-05-29:
Gazebo-classic has a [WideAngleCamera](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/sensors/WideAngleCameraSensor.cc) sensor which uses custom shaders to created a fisheye effect. It should be ported to Ignition eventually. Most of the implementation will probably live in `ign-rendering`.

scpeters, 2021-11-24:
there are pull requests open to implement this in Garden:

* https://github.com/ignitionrobotics/ign-rendering/pull/490
* https://github.com/ignitionrobotics/ign-sensors/pull/171
* https://github.com/ignitionrobotics/ign-gazebo/pull/1215

iche033, 2023-07-12:
wide angle camera support for ogre2 was added in https://github.com/gazebosim/gz-rendering/pull/733. It's available in Harmonic which is to be released in Sep 2023

closed 2023-07-12 by azeey