gazebosim/gz-sensors #25: Joint force-torque sensor

chapulina, 2020-05-29:
We should port Gazebo-classic's [ForceTorqueSensor](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/sensors/ForceTorqueSensor.cc) to Ignition. That sensor attaches to a joint and publishes force/torque data to a topic. The [Force/Torque Sensor tutorial](http://gazebosim.org/tutorials?tut=force_torque_sensor&cat=sensors) provides a lot of context.

nlamprian, 2021-01-08:
I thought I'd give this a go, but I need some help configuring the workspace, given that the sdf class for the force-torque sensor is in sdformat11.

I assume I target Edifice. I check out the relevant versions, or main if the branches do not exist yet. Then the sdformat dependencies (in ign-sensors, ign-physics, and ign-gazebo) must be bumped up. At this point I get errors

```
/home/nlamprian/ignition_master_ws/src/ign-gazebo/src/Conversions.cc:284:52: error: ‘const class sdf::v11::Material’ has no member named ‘FilePath’
  284 |           asFullPath(workflow->MetalnessMap(), _in.FilePath()));
```

which means sdformat master needs to be updated (d03dba22e).

Am I approaching this correctly? And who deals with it?

chapulina, 2021-01-08:
> the sdf class for the force-torque sensor is in sdformat11.
> I assume I target Edifice.

Yeah that's the eventual correct combination, but as you noticed, we're currently pending the upgrade to SDF11 for the entire stack, tracked here: https://github.com/ignition-tooling/release-tools/issues/360. 

Until then, the quickest solution could be to backport https://github.com/osrf/sdformat/pull/393/ to `sdf10` so that you can work with that until we bump to SDF11. CC @azeey

chapulina, 2021-01-20:
> we're currently pending the upgrade to SDF11 for the entire stack

The `main` branch is already depending on SDF 11

keyan, 2021-03-09:
Now that https://github.com/osrf/sdformat/pull/393 has been merged, what work is remaining to resolve this issue?

chapulina, 2021-03-09:
> what work is remaining to resolve this issue?

I think it's useful to look at how other sensors are implemented throughout the stack. For example, see the [IMU](https://github.com/search?q=org%3Aignitionrobotics+imu&type=code). You should need to:

* Add a `ForceTorqueSensor` to `ign-sensors`
* On `ign-gazebo`:
    * Add a `ForceTorqueSensor` component
    * Add a `force_torque_sensor_system`

Also add the necessary code to hook it all up, as well as tests.

diegoferigo, 2021-03-26:
Is there any missing physics feature in the DART plugin that blocks the support of F/T sensors?

chapulina, 2021-04-08:
Leaving a cross reference to https://github.com/ignitionrobotics/ign-physics/pull/143, which seems to be exposing the necessary data from DART through `ign-physics`, but hasn't been merged because it's computationally expensive.

vatanaksoytezer, 2021-05-26:
@chapulina to my understanding this is currently blocked by https://github.com/ignitionrobotics/ign-physics/pull/143. We want to use F/T sensors in ignition, and willing to contribute to get them ported. Can you give some key points / a roadmap for us to look to get this ported?

chapulina, 2021-06-15:
> Can you give some key points / a roadmap for us to look to get this ported?

We're currently planning to evaluate the DART implementation on Gazebo-classic, which is faster, but less precise, see discussion in https://github.com/ignitionrobotics/ign-physics/issues/124 and https://github.com/osrf/gazebo/issues/1321. If that turns out to be unsatisfactory, we'll keep iterating on https://github.com/ignitionrobotics/ign-physics/pull/143.

chapulina, 2021-09-02:
Open PRs related to this:

* https://github.com/ignitionrobotics/ign-gazebo/pull/977
* https://github.com/ignitionrobotics/ign-gazebo/pull/989
* https://github.com/ignitionrobotics/ign-physics/pull/283
* https://github.com/ignitionrobotics/ign-physics/pull/143
* https://github.com/ignitionrobotics/ign-gazebo/pull/952

rickstaa, 2021-12-01:
Judging from the comments above, it seems that the force-torque sensor is not yet available in the ignition. Is this still the case, or has the `force_torque sensor already been released? I saw that it is included in both the [v6.0.0](https://github.com/ignitionrobotics/ign-sensors/releases/tag/ignition-sensors6_6.0.0) source and binary releases of Oct 2, but I am unable to find the right syntax to use it. I have tried to use the syntaxes to include the plugin:

```xml
<plugin filename="libignition-sensors6-force_torque" name="ignition::sensors::ForceTorqueSensor">
</plugin>
```

```xml
<plugin filename="libignition-sensors6-force_torque.so" name="ignition::sensors::ForceTorqueSensor">
</plugin>
```

```xml
<plugin filename="libignition-sensors6-force_torque" name="ignition::sensors::ForceTorqueSensorPrivate">
</plugin>
```

```xml
<plugin filename="libignition-sensors6-force_torque.so" name="ignition::sensors::ForceTorqueSensorPrivate">
</plugin>
```

With all of them I, however keep receiving the following error:

```bash
[Err] [SystemLoader.cc:66] Failed to load system plugin [ignition-gazebo-force-torque-system] : couldn't find shared library.
```

I have installed the binary of ignition-fortress according to [the documentation](https://ignitionrobotics.org/docs/dome/install_ubuntu).

vatanaksoytezer, 2021-12-01:
I have been using FT sensors since 6.0.0 without any problems, with source and binary. Here is what it looks like in robot xacro:

```
  <gazebo reference="${prefix_joint">
    <sensor name="force_torque_sensor" type="force_torque">
      <always_on>true</always_on>
      <update_rate>50</update_rate>
      <visualize>true</visualize>
      <topic>${prefix}_joint/force_torque</topic>
      <force_torque>
        <frame>sensor</frame>
        <measure_direction>parent_to_child</measure_direction>
      </force_torque>
    </sensor>
  </gazebo>
```

and I have the following added in the world (sdf) file:

```
<!-- FT Plugin -->
<plugin
  filename="ignition-gazebo-forcetorque-system"
  name="ignition::gazebo::systems::ForceTorque">
</plugin>
```

If these don't work please make sure and verify you are using version number greater than 6 with `ign gazebo --versions`

rickstaa, 2021-12-01:
@vatanaksoytezer Thanks for the fast response! It was indeed a naming problem. I suspected the shared library name to be equal to the one shown when running the `ldconfig -p | grep libignition-sensors6-force` command (i.e. `libignition-sensors6-force_torque.so`). I think the proposal made in #103 would remove this confusion.

chapulina, 2021-12-03:
Oh we can close this!

closed 2021-12-03 by chapulina

vbalaji21, 2024-05-16:
Does the visualisation work in F/T sensor ? I am trying it and unable to see it

PR #144 (2021-09-22): Joint Force-Torque Sensor
# 🎉 New feature

Closes #25 

## Summary

Joint force-torque sensor implementation.

## Test it

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://ignitionrobotics.org/docs/all/contributing#test-coverage))
- [ ] While waiting for a review on your PR, please help review [ano [... cut]