gazebosim/gz-sensors #26: CPU lidar (no rendering)

chapulina, 2020-05-29:
Ignition already supports [GPU Lidars](https://github.com/ignitionrobotics/ign-sensors/blob/master/src/GpuLidarSensor.cc), which use `ign-rendering` and detect visuals in the scene.

Besides the GPU version of this sensor, Gazebo-classic also supported a CPU-based [ray sensor](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/sensors/RaySensor.cc), which used the physics engine to detect objects. Even though that sensor is less performant than the GPU version, over the years many users have relied on it to have a ray sensor on settings where using a GPU / rendering engine was not ideal, like on cloud machines.

Based on this past experience, we should also consider adding a render-less lidar sensor to Ignition.

caioaamaral, 2021-04-06:
@chapulina Seems like the hard work is in the implementation of a `MultiRayShape` an then somehow instantiate it inside a `CpuLidar` sensor plugin.

 What is the ignition equivalent of calling `gazebo::physics::PhysicsEngine::CreateCollision(...)`?

chapulina, 2021-04-28:
One way that I can think of this sensor being implemented is:

* Adding a `MultiRayShape` to `ign-physics` similar to the existing shapes, i.e. [CapsuleShape](https://github.com/ignitionrobotics/ign-physics/blob/ign-physics4/include/ignition/physics/CapsuleShape.hh)
* Adding the implementation for DART
* Create a component in `ign-gazebo` to hold Lidar data
* On [ignition::gazebo::systems::Physics](https://github.com/ignitionrobotics/ign-gazebo/blob/f795b657683f629133e7de056771959786a150fd/src/systems/physics/Physics.cc#L1958-L1970), populate that data similar to how we populate `WorldLinearAcceleration`.
* Create a `CpuLidar` sensor here
* Create a `CpiuLidar` Gazebo system similar to [systems::Imu](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/src/systems/imu/Imu.cc) that instantiates `sensors::CpuLidar` and updates it with the component data

azeey, 2022-11-10:
In the meantime, a workaround for those who need to simulate lidars in an environment where a GPU is not available is to use `gpu_lidar` but with software rendering. Running `gz-sim` as is with a `gpu_lidar` in such an environment would currently cause a crash with an error: `[GUI] [Err] [Ogre2RenderEngine.cc:1098]  Unable to create the rendering window: ...`. The workaround is to set the environment variable `MESA_GL_VERSION_OVERRIDE=3.3`.

This is also how we have setup some of our CI machines (eg. Github Actions). There, since there is also no X server, we start one using `Xvfb`:

```
Xvfb :1 -noreset  -screen 0 1280x1024x24 &
export DISPLAY=:1.0
export MESA_GL_VERSION_OVERRIDE=3.3
```

romulogcerqueira, 2024-05-06:
Hi all,

it seems that DART does support ray collision via [Bullet backend](https://github.com/dartsim/dart/discussions/1803), which is [also supported](https://github.com/gazebosim/gz-physics/blob/d0a6e507210cb17f39a4d615fcdf75adfeda4838/dartsim/src/WorldFeatures.cc#L47-L50) by `gz-physics` (being configured through SDF [<collision_dectector>](http://sdformat.org/spec?ver=1.11&elem=physics#dart_collision_detector)).

This way, one high-level implementation proposal for the ray-based collisions would be:

- [`gz-physics`]
    - Create `GetRaycasts` (similar to [`GetContacts`](https://github.com/gazebosim/gz-physics/blob/gz-physics7/include/gz/physics/GetContacts.hh))
        - Create structs to store ray information (`Ray`, `RaycastResult` and `RayHit`)
        - Create `GetRaycastsFromLastStepFeature` simulation feature
        - Create `GetRaycastsFromLastStep()` to get the raycast results from last simulation step
    - Incorporate this feature into [`gz::physics::dartsim::SimulationFeatures`](https://github.com/gazebosim/gz-physics/blob/gz-physics7/dartsim/src/SimulationFeatures.hh)
- [`gz-sensors`]
    - Create sensor `RaySensor` here
- [`gz-sim`]
    - Create component `RaycastResult` to hold raycast data
    - Create `UpdateRaycastCollisions()` into [`Physics.cc`](https://github.com/gazebosim/gz-sim/blob/gz-sim8/src/systems/physics/Physics.cc)
        - Call `GetRaycastsFromLastStep()` feature to get the raycast results
        - Populate `RaycastResult` component data with output of this feature
        - Write component to ECM
    - Create system plugin `Raycast` that instanciates `RaySensor` and updates it with component data from ECM.

Any suggestions/feedback?

ggari-robotnik, 2024-07-11:
I'm trying to run 2D laser simulations without GPU inside a container using this launch:
```bash
ros2 launch ros_gz_sim_demos gpu_lidar_bridge.launch.py
```

But the 2D laser scans and 3D beams do not update. (I checked that simulation is not paused).  

The workaround proposed by @azeey do not work for me. Any idea how make it work?

azeey, 2024-07-11:
Do you see any errors when launching? Can you try running https://github.com/gazebosim/gz-sim/blob/b8bbf1d9dee47af68fd1c85fa8382afc47732ff2/examples/worlds/gpu_lidar_sensor.sdf in Gazebo and with `-v4` to get debug output?

ggari-robotnik, 2024-07-12:
Here you are the output, But v4 do not seem to work, I think I'm doing something wrong about that, but I cannot figure it out:
```
ros2 launch ros_gz_sim_demos gpu_lidar_bridge.launch.py  gz_args:=v4 
[INFO] [launch]: All log files can be found below /home/robot/.ros/log/2024-07-12-10-09-33-274163-rob-id-002-2648
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [2651]
[INFO] [parameter_bridge-2]: process started with pid [2653]
[INFO] [rviz2-3]: process started with pid [2656]
[rviz2-3] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-robot'
[parameter_bridge-2] [INFO] [1720771773.350842014] [ros_gz_bridge]: Creating GZ->ROS Bridge: [lidar (gz.msgs.LaserScan) -> lidar (sensor_msgs/msg/LaserScan)] (Lazy 0)
[parameter_bridge-2] [INFO] [1720771773.352148142] [ros_gz_bridge]: Creating ROS->GZ Bridge: [lidar (sensor_msgs/msg/LaserScan) -> lidar (gz.msgs.LaserScan)] (Lazy 0)
[parameter_bridge-2] [INFO] [1720771773.352544802] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/lidar/points (gz.msgs.PointCloudPacked) -> /lidar/points (sensor_msgs/msg/PointCloud2)] (Lazy 0)
[parameter_bridge-2] [INFO] [1720771773.352772269] [ros_gz_bridge]: Creating ROS->GZ Bridge: [/lidar/points (sensor_msgs/msg/PointCloud2) -> /lidar/points (gz.msgs.PointCloudPacked)] (Lazy 0)
[ruby $(which ign) gazebo-1] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-robot'
```

[image]
[image]

azeey, 2024-07-12:
Oh, I meant directly using `gz sim -v4` to isolate the issue to Gazebo.

Anyway, I was able to reproduce this issue. It's happening because the `UserCommands` plugin is missing. When you translated the box, it gave the appearance that it translated, but it didn't actually get translated. The GUI sends calls the translate service, but since `UserCommands` is missing, the service is not available. If you click off, you'd see the box would snap back to its original position. 

I think we have two action items:
1. The translate tool should either be disabled if the UserCommands plugin is not available or request for the plugin to be added
2. The gpu_lidar_sensor.sdf example should include the UserCommands plugin.

I'll go ahead and create issues for these. Any help to fix them would be greatly appreciated.

VoidDruid, 2024-08-22:
Hi all! I am running gazebo in a VM with GPU passthrough, I got everything to work with `--render-engine ogre` (world rendering, camera sensor) except for LiDAR. The workaround with software rendering works, but significantly reduces simulation performance.

Since in my case GPU not being used for lidar is probably a virtualization problem and I don't have the expertise to fix it, an acceptable workaround would be to use GPU rendering and CPU lidar. But from looking at the issues and PRs mentioned above by @azeey, I understand that CPU lidar is not ready yet. Will it be included in future releases or will it come as an update? (I use gz harmonic)

romulogcerqueira, 2025-02-12:
Hi @azeey,

thanks for checking in! The last item is still around, however, it also needs contributions to `gz-sensors` on creating a ray-based collision sensor (CpuLidarSensor?).

I’ll revisit it when we have more free cycles. I’d also be happy to guide anyone who wants to take it on.

dadaroce, 2025-06-13:
Hi @romulogcerqueira,

I’d be interested in contributing to the ray-based collision sensor (CpuLidarSensor) in `gz-sensors`. If you have any design discussions, open issues, or partial implementations to reference, I’d be glad to review and pick it up.

romulogcerqueira, 2025-06-25:
> Hi [@romulogcerqueira](https://github.com/romulogcerqueira),
> 
> I’d be interested in contributing to the ray-based collision sensor (CpuLidarSensor) in `gz-sensors`. If you have any design discussions, open issues, or partial implementations to reference, I’d be glad to review and pick it up.

Hi @dadaroce,

Great to hear you're interested in contributing!

I haven't started this development yet, however, what I had in mind was to create a `CpuLidarSensor` here (similar to [GpuLidarSensor](https://github.com/gazebosim/gz-sensors/blob/gz-sensors9/include/gz/sensors/GpuLidarSensor.hh), but extending from the `Sensor` class instead of `Lidar` one, which depends on `gz-rendering`) to handle raycasts based on physical collisions.

azeey, 2026-02-26:
We're still in the process of updating `main` branches right now (see https://github.com/gazebo-tooling/release-tools/issues/1403), so the regular PR CI will be broken for some time. In the mean time, we can run manual CI jobs that builds all the branches together. 

Don't worry about the `jetty` label. It's not used by CI at all. I'll add the `kura` label on them manually.