gazebosim/gz-sensors #28: Wireless transmitter and receiver

chapulina, 2020-05-30:
Gazebo-classic has 3 wireless sensors that worked together:
* [WirelessReceiver](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/sensors/WirelessReceiver.cc) 
* [WirelessTransmitter](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/sensors/WirelessTransmitter.cc) 
* [WirelessTransceiver](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/sensors/WirelessTransceiver.cc) 

It uses the Hata-Okumara model for the signal propagation and obstacles between transmitter and receiver are taken into account. For more context, see [the original pull request](https://osrf-migration.github.io/gazebo-gh-pages/#!/osrf/gazebo/pull-requests/644/page/1).

They should eventually be ported to Ignition if there are interested users. I don't think these sensors are very widely used.

denper19, 2023-08-06:
Hello, is this something you would be OK with a PR on? It will be my first time and I would love to try contributing.

mjcarroll, 2023-08-06:
@denper19 While it wasn't updated here, there is now a comms model inside of gz-sim.  You can use it to simulate "perfect comms" (no channel loss) https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/perfect_comms/PerfectComms.cc or something lossy, like RF comms: https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/rf_comms/RFComms.cc

There is an example that exercises this here: https://github.com/gazebosim/gz-sim/blob/gz-sim7/examples/worlds/rf_comms.sdf

iche033, 2023-08-07:
closing this as the feature is now available in gz-sim.

closed 2023-08-07 by iche033