gazebosim/gz-sensors #40: Unable to load igintion-transport9

eladpar, 2020-08-25:
after Installing ign-transport9 from tutorial
 (
installing from source because when installing 
`sudo apt-get install libignition-transport9-dev`
i get an error
`Unable to locate package libignition-transport9-dev`
)
when trying to cmake ign-sensors i get:
`
  Could not find a package configuration file provided by
  "ignition-transport9" with any of the following names:

    ignition-transport9Config.cmake
    ignition-transport9-config.cmake
`

iche033, 2020-08-25:
ignition-transport9 has not been released yet so you'll need to install from source. 

I recommend following the instructions here for installing the whole ignition stack, either from source or debs: https://ignitionrobotics.org/docs/citadel

chapulina, 2020-08-25:
Closing since the version is expected not to exist.

Just curious @eladpar , howcome you were trying to install that? Did you see instructions somewhere?

closed 2020-08-25 by chapulina

eladpar, 2020-08-26:
> Closing since the version is expected not to exist.
> 
> Just curious @eladpar , howcome you were trying to install that? Did you see instructions somewhere?

yes .
after trying to install from Source ign-sensors
one of the prerequisites is ign-transpot -> instaalation Guide ->
https://ignitionrobotics.org/api/transport/9.0/installation.html
I guess the installation guide is corrupt