gazebosim/gz-sensors #43: Noise not being applied to the Points publisher of GpuLidar sensor

eric1221bday, 2020-08-28:
On a high level this can be replicated by running the SubT virtual course. The command used is:
```
ign launch -v 4 tunnel_circuit_practice.ign robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_3
```
`X1_SENSOR_CONFIG_3` is supposed to contain a 3D Lidar, and the output looks like
[image]

There's clearly no noise even though the stddev defined in the sdf specifies 0.01 noise. This does not occur for the 2D Lidars which output a LaserScan message instead. GpuLidar internally publishes both LaserScan and PointCloud messages, via the `GpuLidarSensor::Update` method which calls `this->PublishLidarScan(_now);` from the Lidar base class. 

The issue here is that the noise is only applied to the range data in the Lidar base class, in https://github.com/ignitionrobotics/ign-sensors/blob/8ee625a405264ca697fbefc0e8a9b1ae6d1b53dd/src/Lidar.cc#L229 which is called here https://github.com/ignitionrobotics/ign-sensors/blob/8ee625a405264ca697fbefc0e8a9b1ae6d1b53dd/src/GpuLidarSensor.cc#L233 That interior call doesn't update the laser buffer itself, merely taking a copy here https://github.com/ignitionrobotics/ign-sensors/blob/8ee625a405264ca697fbefc0e8a9b1ae6d1b53dd/src/Lidar.cc#L234 so the actual laser buffer remains unchanged. As a result the outer `GpuLidarSensor::Update` only has access to the original clean lidar ranges, hence why it publishes point clouds with no noise.

The solution here is probably to separate out an `ApplyNoise` function in `Lidar` that will properly alter the `this->laserBuffer` and call it before both `this->PublishLidarScan(_now);` and `this->dataPtr->FillPointCloudMsg();` so it can be accessed by both classes.

ahcorde, 2020-11-04:
Hello 

I tested de GPU lidar with the example world in ign-gazebo, the noise it's being added to the rays. See the following gif:

[image]

The values which contain the maximum range value are not adding the noise. But in your image I think all of them are colliding with something.

iche033, 2020-11-04:
I think the the visualization in ign-gazebo shows lidar scan data which has noise correctly applied but the issue is that the point cloud data that are published over the `~/<topic_name>/points` topic doesn't have any noise applied.

ahcorde, 2020-11-05:
The plugin uses a subscriber to get the data: https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo4/src/gui/plugins/visualize_lidar/VisualizeLidar.cc#L381

acschang, 2021-02-09:
Resolved in #86

closed 2021-11-15 by chapulina