gazebosim/gz-sensors #45: Test Regression from #35

azeey, 2020-09-02:
The following tests have started [failing](https://build.osrfoundation.org/job/ignition_sensors-ci-ign-sensors2-bionic-amd64/36/) since #35 
* DepthCameraSensor/DepthCameraSensorTest.ImagesWithBuiltinSDF/ogre2
   
```
/var/lib/jenkins/workspace/ignition_sensors-ci-ign-sensors2-bionic-amd64/ign-sensors/test/integration/depth_camera_plugin.cc:435
The difference between expectedDepth and d is 0.19999999999999996, which exceeds 1e-6, where
expectedDepth evaluates to 0.19999999999999996,
d evaluates to 0, and
1e-6 evaluates to 9.9999999999999995e-07.
```
* RgbdCameraSensor/RgbdCameraSensorTest.ImagesWithBuiltinSDF/ogre2
```
/var/lib/jenkins/workspace/ignition_sensors-ci-ign-sensors2-bionic-amd64/ign-sensors/test/integration/rgbd_camera_plugin.cc:553
Expected equality of these values:
  -ignition::math::INF_D
    Which is: -inf
  g_depthBuffer[index]
    Which is: 0
```

chapulina, 2020-09-02:
Probably caused by https://github.com/ignitionrobotics/ign-rendering/issues/129

iche033, 2020-09-02:
being addressed in https://github.com/ignitionrobotics/ign-rendering/pull/130

closed 2020-09-28 by iche033