gazebosim/gz-sensors #49: ignition::sensors::SensorFactory.CreateSensor(...) returns null in several tests

jacobperron, 2020-09-22:
It looks like in several plugin tests, calling `ignition::sensors::SensorFactory.CreateSensor(...)` without a template type returns `null`, resulting in several test failures:

https://build.osrfoundation.org/job/ign_sensors-ci-win/80/testReport/

Currently, only seeing these failures on Windows.

jacobperron, 2020-09-22:
Also affecting `ign-sensors3`: https://build.osrfoundation.org/job/ign_sensors-ign-3-win/16/testReport/

chapulina, 2020-09-22:
It looks like the sensor plugins can't be found:

```
23: [Err] [D:\Jenkins\workspace\ign_sensors-ci-win\ws\ign-sensors\src\SensorFactory.cc:74] Unable to find sensor plugin path for [ignition-sensors4-air_pressure]
23: [Err] [D:\Jenkins\workspace\ign_sensors-ci-win\ws\ign-sensors\src\SensorFactory.cc:162] Unable to instantiate sensor plugin for [ignition-sensors4-air_pressure]
23: [Err] [SensorFactory.hh:136] Failed to create sensor of type[air_pressure]
```

On the change to use  `ign-plugin` in https://github.com/ignitionrobotics/ign-sensors/pull/38, it seems that @ahcorde fixed plugin loading on Windows. I haven't looked closely at the fix and don't know if it is backportable. That PR targets Ignition-E.

jacobperron, 2020-09-22:
Also affecting `ign-sensors2`: https://build.osrfoundation.org/job/ign_sensors-ign-2-win/42

scpeters, 2020-11-10:
this should be fixed by #60

scpeters, 2020-11-17:
fixed by #60

closed 2020-11-17 by scpeters