gazebosim/gz-sensors #62: DRI integration tests fail on macOS due to duplicate ogre1.9/ogre2.1 symbols

scpeters, 2020-11-11:
After fixing many macOS tests in #60, there are still some remaining test failures that seem to be caused by duplicate symbols in the ogre 1.9 and 2.1 libraries. Here is some example console output:

~~~
13: [ RUN      ] CameraSensor/CameraSensorTest.ImagesWithBuiltinSDF/ogre2
13: objc[24379]: Class OgreConfigWindowDelegate is implemented in both /usr/local/opt/ogre1.9/lib/libOgreMain.1.9.0.dylib (0x107ec5dc0) and /usr/local/opt/ogre2.1/lib/libOgreMain.2.1.0.dylib (0x1078ee838). One of the two will be used. Which one is undefined.
13: dyld: lazy symbol binding failed: Symbol not found: __ZN4Ogre2v113OverlaySystemC1Ev
13:   Referenced from: /usr/local/Cellar/ignition-rendering4/4.0.0/lib/libignition-rendering4-ogre2.dylib
13:   Expected in: flat namespace
13: 
13: dyld: Symbol not found: __ZN4Ogre2v113OverlaySystemC1Ev
13:   Referenced from: /usr/local/Cellar/ignition-rendering4/4.0.0/lib/libignition-rendering4-ogre2.dylib
13:   Expected in: flat namespace
13: 
13/32 Test #13: INTEGRATION_camera_plugin ...................Child aborted***Exception:   0.23 sec
~~~

Here is a backtrace that pinpoints the `SIGABRT` signal to [camera_plugin.cc:54](https://github.com/ignitionrobotics/ign-sensors/blob/ignition-sensors4_4.0.0/test/integration/camera_plugin.cc#L54) where `ignition::rendering::engine(std::string)` is called.

~~~
(lldb) target create "bin/INTEGRATION_camera_plugin"
Current executable set to 'bin/INTEGRATION_camera_plugin' (x86_64).
(lldb) r
Process 1162 launched: '/Users/scpeters/ws/sensors/build/ignition-sensors4/bin/INTEGRATION_camera_plugin' (x86_64)
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from CameraSensor/CameraSensorTest
[ RUN      ] CameraSensor/CameraSensorTest.ImagesWithBuiltinSDF/ogre2
objc[1162]: Class OgreConfigWindowDelegate is implemented in both /usr/local/opt/ogre1.9/lib/libOgreMain.1.9.0.dylib (0x104fd9dc0) and /usr/local/opt/ogre2.1/lib/libOgreMain.2.1.0.dylib (0x104ad5838). One of the two will be used. Which one is undefined.
dyld: lazy symbol binding failed: Symbol not found: __ZN4Ogre2v113OverlaySystemC1Ev
  Referenced from: /usr/local/Cellar/ignition-rendering4/4.0.0/lib/libignition-rendering4-ogre2.dylib
  Expected in: flat namespace

dyld: Symbol not found: __ZN4Ogre2v113OverlaySystemC1Ev
  Referenced from: /usr/local/Cellar/ignition-rendering4/4.0.0/lib/libignition-rendering4-ogre2.dylib
  Expected in: flat namespace

Process 1162 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00000001000f73ba dyld`__abort_with_payload + 10
dyld`__abort_with_payload:
->  0x1000f73ba <+10>: jae    0x1000f73c4               ; <+20>
    0x1000f73bc <+12>: movq   %rax, %rdi
    0x1000f73bf <+15>: jmp    0x1000f69c9               ; cerror_nocancel
    0x1000f73c4 <+20>: retq   
Target 0: (INTEGRATION_camera_plugin) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00000001000f73ba dyld`__abort_with_payload + 10
    frame #1: 0x00000001000f6bac dyld`abort_with_payload_wrapper_internal + 82
    frame #2: 0x00000001000f6bde dyld`abort_with_payload + 9
    frame #3: 0x00000001000b6a9d dyld`dyld::halt(char const*) + 343
    frame #4: 0x00000001000b6bc7 dyld`dyld::fastBindLazySymbol(ImageLoader**, unsigned long) + 167
    frame #5: 0x00007fff5944b32e libdyld.dylib`dyld_stub_binder + 282
    frame #6: 0x000000010399d008 libignition-rendering4-ogre2.dylib
    frame #7: 0x00000001039617bf libignition-rendering4-ogre2.dylib`ignition::rendering::v4::Ogre2RenderEngine::LoadAttempt() + 37
    frame #8: 0x00000001039614fd libignition-rendering4-ogre2.dylib`ignition::rendering::v4::Ogre2RenderEngine::LoadImpl(std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > const&) + 377
    frame #9: 0x00000001004eb929 libignition-rendering4.4.dylib`ignition::rendering::v4::BaseRenderEngine::Load(std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > const&) + 233
    frame #10: 0x00000001004df056 libignition-rendering4.4.dylib`ignition::rendering::RenderEngineManagerPrivate::Engine(std::__1::__map_iterator<std::__1::__tree_iterator<std::__1::__value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, ignition::rendering::v4::RenderEngine*>, std::__1::__tree_node<std::__1::__value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, ignition::rendering::v4::RenderEngine*>, void*>*, long> >, std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 58
    frame #11: 0x00000001004dee9e libignition-rendering4.4.dylib`ignition::rendering::v4::RenderEngineManager::Engine(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 84
    frame #12: 0x0000000100008091 INTEGRATION_camera_plugin`CameraSensorTest::ImagesWithBuiltinSDF(this=<unavailable>, _renderEngine="ogre2") at camera_plugin.cc:54:18 [opt]
    frame #13: 0x000000010000a9d7 INTEGRATION_camera_plugin`CameraSensorTest_ImagesWithBuiltinSDF_Test::TestBody(this=<unavailable>) at camera_plugin.cc:92:3 [opt]
    frame #14: 0x000000010002abd8 INTEGRATION_camera_plugin`void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) [inlined] void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(method=<unavailable>, location=<unavailable>)(), char const*) at gtest.cc:2445:10 [opt]
    frame #15: 0x000000010002abc9 INTEGRATION_camera_plugin`void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(object=<unavailable>, method=<unavailable>, location="the test body")(), char const*) at gtest.cc:2481 [opt]
    frame #16: 0x000000010002aad1 INTEGRATION_camera_plugin`testing::Test::Run(this=0x00000001020a4d20) at gtest.cc:2519:5 [opt]
    frame #17: 0x000000010002beec INTEGRATION_camera_plugin`testing::TestInfo::Run(this=0x00000001020a5460) at gtest.cc:2695:11 [opt]
    frame #18: 0x000000010002c8b7 INTEGRATION_camera_plugin`testing::TestCase::Run(this=0x00000001020a5710) at gtest.cc:2813:28 [opt]
    frame #19: 0x00000001000395b7 INTEGRATION_camera_plugin`testing::internal::UnitTestImpl::RunAllTests(this=<unavailable>) at gtest.cc:5179:43 [opt]
[... cut ...]
    frame #22: 0x0000000100038fdc INTEGRATION_camera_plugin`testing::UnitTest::Run(this=0x0000000100059c88) at gtest.cc:4788:10 [opt]
    frame #23: 0x000000010000af31 INTEGRATION_camera_plugin`main [inlined] RUN_ALL_TESTS() at gtest.h:2341:46 [opt]
    frame #24: 0x000000010000af24 INTEGRATION_camera_plugin`main(argc=1, argv=<unavailable>) at camera_plugin.cc:102 [opt]
    frame #25: 0x00007fff5945f3d5 libdyld.dylib`start + 1
    frame #26: 0x00007fff5945f3d5 libdyld.dylib`start + 1
(lldb)
~~~

chapulina, 2020-11-11:
> duplicate symbols in the ogre 1.9 and 2.1 libraries

This may be related to https://github.com/ignitionrobotics/ign-rendering/issues/36, where Ogre 1 and 2 can't be loaded at the same time.

I'm wondering why this would happen here though, since the tests should only be using one or the other. Preferably `ogre2`, since that's the default.

iche033, 2020-11-11:
I think that happens when both the ogre 1.x and 2.x's libOgreMain.so / dylib are in the library path, in which case the wrong library version gets loaded. I ran into this when I tried to run ign-rendering's ogre2 tests and had ogre 1.x also in my colcon workspace. The test then loads the wrong libOgreMain.so library.

scpeters, 2020-11-12:
I've noticed that the `ignition-rendering4` bottle links against the `ogre1.9` version of the `libOgreOverlay` library:

~~~
otool -L /usr/local/opt/ignition-rendering4/lib/libignition-rendering4-ogre2.dylib 
/usr/local/opt/ignition-rendering4/lib/libignition-rendering4-ogre2.dylib:
	/usr/local/opt/ignition-rendering4/lib/libignition-rendering4-ogre2.4.dylib (compatibility version 4.0.0, current version 4.1.0)
	@rpath/libignition-rendering4.4.dylib (compatibility version 4.0.0, current version 4.1.0)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/ogre2.1/lib/libOgreMain.2.1.0.dylib (compatibility version 2.1.0, current version 2.1.0)
	/usr/local/opt/ignition-common3/lib/libignition-common3-graphics.3.dylib (compatibility version 3.0.0, current version 3.7.0)
	/usr/local/opt/ignition-common3/lib/libignition-common3-events.3.dylib (compatibility version 3.0.0, current version 3.7.0)
	/usr/local/opt/ignition-common3/lib/libignition-common3.3.dylib (compatibility version 3.0.0, current version 3.7.0)
	/usr/local/opt/ossp-uuid/lib/libuuid.16.dylib (compatibility version 17.0.0, current version 17.22.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.0.0)
	/usr/local/opt/ignition-math6/lib/libignition-math6.6.dylib (compatibility version 6.0.0, current version 6.6.0)
	/usr/local/opt/ignition-plugin1/lib/libignition-plugin1.1.dylib (compatibility version 1.0.0, current version 1.1.0)
	/usr/local/opt/ogre2.1/lib/libOgreHlmsPbs.2.1.0.dylib (compatibility version 2.1.0, current version 2.1.0)
	/usr/local/opt/ogre2.1/lib/libOgreHlmsUnlit.2.1.0.dylib (compatibility version 2.1.0, current version 2.1.0)
	/usr/local/opt/ogre1.9/lib/libOgreOverlay.1.9.0.dylib (compatibility version 1.9.0, current version 1.9.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 800.7.0)
~~~

If I build from source and install, then it is linked properly. This may be caused by homebrew somehow.

scpeters, 2020-11-13:
Through some debugging, I noticed that when building `ignition-rendering` with `brew`, the `CMAKE_PREFIX_PATH` environment variable includes `/usr/local`, which causes the `/usr/local/lib/libOgreOverlay.dylib` symbolic link to the ogre1.9 library to be preferred (since `CMAKE_PREFIX_PATH` is evaluated before `HINTS` in the `find_library` call used in `FindIgnOGRE2`). I believe https://github.com/ignitionrobotics/ign-cmake/pull/125 will fix it.

closed 2020-11-16 by scpeters