gazebosim/gz-sensors #66: GpuLidarSensorTest.VerticalLidar/ogre2 failure on macOS

scpeters, 2020-11-17:
Several tests on macOS were failing to load the `ignition-rendering*-ogre2` plugin (#62), which has been fixed, but now the [GpuLidarSensorTest.VerticalLidar/ogre2](https://build.osrfoundation.org/job/ignition_sensors-ci-ign-sensors4-homebrew-amd64/4/testReport/junit/(root)/GpuLidarSensor_GpuLidarSensorTest/VerticalLidar_ogre2/) test is failing:

~~~
test/integration/gpu_lidar_sensor_plugin.cc:635
The difference between sensor->Range(i * horzSamples + mid) and expectedRange is inf, which exceeds 1e-3, where
sensor->Range(i * horzSamples + mid) evaluates to inf,
expectedRange evaluates to 0.35355339059327368, and
1e-3 evaluates to 0.001.
~~~

chapulina, 2021-02-19:
Since #101 backported the fix to `ign-sensors3`, besides the test above, there's also these 2:

```
17: [ RUN      ] GpuLidarSensor/GpuLidarSensorTest.TestThreeBoxes/ogre2
17: [Err] [Ogre2RenderEngine.cc:432] Unable to find Ogre Plugin[/usr/local/opt/ogre2.1/lib/OGRE-2.1/RenderSystem_GL3Plus]. Rendering will not be possible.Make sure you have installed OGRE properly.
17: [Msg] Loading plugin [ignition-rendering-ogre2]
17: [Msg] Publishing laser scans on [/ignition/sensors/test/lidar1]
17: [Msg] Publishing laser scans on [/ignition/sensors/test/lidar2]
17: /Users/jenkins/workspace/ignition_sensors-ci-ign-sensors3-homebrew-amd64/ign-sensors/test/integration/gpu_lidar_sensor_plugin.cc:519: Failure
17: Expected equality of these values:
17:   sensor1->Range(last)
17:     Which is: 4.5000276565551758
17:   ignition::math::INF_D
17:     Which is: inf
17: [  FAILED  ] GpuLidarSensor/GpuLidarSensorTest.TestThreeBoxes/ogre2, where GetParam() = "ogre2" (300 ms)
17: [ RUN      ] GpuLidarSensor/GpuLidarSensorTest.VerticalLidar/ogre2
17: [Err] [Ogre2RenderEngine.cc:432] Unable to find Ogre Plugin[/usr/local/opt/ogre2.1/lib/OGRE-2.1/RenderSystem_GL3Plus]. Rendering will not be possible.Make sure you have installed OGRE properly.
17: [Msg] Loading plugin [ignition-rendering-ogre2]
17: [Msg] Publishing laser scans on [/ignition/sensors/test/lidar]
17: /Users/jenkins/workspace/ignition_sensors-ci-ign-sensors3-homebrew-amd64/ign-sensors/test/integration/gpu_lidar_sensor_plugin.cc:634: Failure
17: The difference between sensor->Range(i * horzSamples + mid) and expectedRange is inf, which exceeds 1e-3, where
17: sensor->Range(i * horzSamples + mid) evaluates to inf,
17: expectedRange evaluates to 0.35355339059327368, and
17: 1e-3 evaluates to 0.001.
17: [  FAILED  ] GpuLidarSensor/GpuLidarSensorTest.VerticalLidar/ogre2, where GetParam() = "ogre2" (240 ms)
```

darksylinc, 2021-06-06:
I ran into this problem on Linux as well and it seems that it will be fixed with ignitionrobotics/ign-rendering#333

darksylinc, 2021-06-14:
The PR has been merged into ign-rendering. It may be worth trying this test again to see if it's fixed on macOS

iche033, 2021-06-14:
since the issue is reported for ign-sensors3, I backported the macOS fixes to ign-rendering3 in https://github.com/ignitionrobotics/ign-rendering/pull/339. If that looks good, we can bump the ign-rendering3 homebrew bottle version then test to see if it fixes the ign-sensors3 tests.