gazebosim/gz-sensors #67: RgbdCameraSensorTest.ImagesWithBuiltinSDF/ogre2 failure on macOS

scpeters, 2020-11-17:
Several tests on macOS were failing to load the `ignition-rendering*-ogre2` plugin (#62), which has been fixed, but now the [RgbdCameraSensorTest.ImagesWithBuiltinSDF/ogre2](https://build.osrfoundation.org/job/ignition_sensors-ci-ign-sensors4-homebrew-amd64/4/testReport/junit/(root)/RgbdCameraSensor_RgbdCameraSensorTest/ImagesWithBuiltinSDF_ogre2/) test is failing:

~~~
19: test/integration/rgbd_camera_plugin.cc:354: Failure
19: Expected: (mb) > (0u), actual: 0 vs 0
19: test/integration/rgbd_camera_plugin.cc:417: Failure
19: Expected: (mb) > (0u), actual: 0 vs 0
~~~

scpeters, 2020-11-18:
I added the following code to the test to save a received image to a png file to help debug the macOS failure:

~~~
diff --git a/test/integration/rgbd_camera_plugin.cc b/test/integration/rgbd_camera_plugin.cc
index 8175560..e91deb9 100644
--- a/test/integration/rgbd_camera_plugin.cc
+++ b/test/integration/rgbd_camera_plugin.cc
@@ -20,6 +20,7 @@
 
 #include <ignition/common/Filesystem.hh>
 #include <ignition/common/Event.hh>
+#include <ignition/common/Image.hh>
 #include <ignition/sensors/Manager.hh>
 #include <ignition/sensors/RgbdCameraSensor.hh>
 #include <ignition/rendering.hh>
@@ -317,6 +318,13 @@ void RgbdCameraSensorTest::ImagesWithBuiltinSDF(
   imgCounter = 0;
   pcCounter = 0;
 
+  ignition::common::Image image;
+  image.SetFromData(g_imgBuffer,
+                    rgbdSensor->ImageWidth(),
+                    rgbdSensor->ImageHeight(),
+                    ignition::common::Image::RGB_INT8);
+  image.SavePNG("saved_image.png");
+
   // depth image indices
   int midWidth = rgbdSensor->ImageWidth() * 0.5;
   int midHeight = rgbdSensor->ImageHeight() * 0.5;
~~~

## Image from successful Linux test

[image]

## Image from unsuccessful macOS test

[image]

scpeters, 2020-11-18:
here's the ogre2.log file from macOS: [rgbd_macos_ogre2.log](https://github.com/ignitionrobotics/ign-sensors/files/5558407/rgbd_macos_ogre2.log)

scpeters, 2020-11-18:
I'm comparing the ogre2 log from macOS with a log from a successful run on my Linux machine

the following line appears in both logs:

~~~
Can't assign material scene::Material(65526) because this Material does not exist. Have you forgotten to define it in a .material script?
~~~

but the following lines appear only on macOS:

~~~
23:22:26: Vertex Shader: 536969216VertexShader_vs
Fragment Shader: 536969216PixelShader_ps
 GLSL link result : 
WARNING: Could not find vertex shader attribute 'blendWeights' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv5' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'secondary_colour' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv1' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv4' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'blendWeights2' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'blendIndices' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv6' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'blendIndices2' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'colour' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv2' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'normal' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'tangent' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv3' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv7' to match BindAttributeLocation request.
23:22:26: Vertex Shader: 536969216VertexShader_vs
Fragment Shader: 536969216PixelShader_ps
 GLSL validation result : 
Validation Failed: Sampler error:
  Samplers of different types use the same texture image unit.
   - or -
  A sampler's texture unit is out of range (greater than max allowed or negative).
23:22:29: Vertex Shader: DepthCameraVS
Fragment Shader: DepthCameraFS
 GLSL link result : 
WARNING: Could not find vertex shader attribute 'qtangent' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'blendWeights' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv5' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'secondary_colour' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv1' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv4' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'blendWeights2' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'blendIndices' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv6' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'blendIndices2' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'colour' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv2' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'tangent' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv3' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv7' to match BindAttributeLocation request.
23:22:29: Vertex Shader: DepthCameraFinalVS
Fragment Shader: DepthCameraFinalFS
 GLSL link result : 
WARNING: Could not find vertex shader attribute 'qtangent' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'blendWeights' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv5' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'secondary_colour' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv1' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv4' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'blendWeights2' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'blendIndices' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv6' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'blendIndices2' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'colour' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv2' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'normal' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'tangent' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv3' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'uv7' to match BindAttributeLocation request.
~~~

iche033, 2020-11-19:
```
 Validation Failed: Sampler error:
  Samplers of different types use the same texture image unit.
   - or -
  A sampler's texture unit is out of range (greater than max allowed or negative).
````

That looks suspicious. The shaders with name that starts with numbers are dynamically generated by ogre. Setting [this line of code](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering4/ogre2/src/Ogre2RenderEngine.cc#L669) in ign-rendering to (true, true), will tell ogre to write the shaders out to file on disk (I think it'll be written to where the executable is run). If you have time, can you give that a try that and share the generated shader files and ogre2.log?

scpeters, 2020-11-19:
> ```
>  Validation Failed: Sampler error:
>   Samplers of different types use the same texture image unit.
>    - or -
>   A sampler's texture unit is out of range (greater than max allowed or negative). [...]

[rgbd_failure.tar.gz](https://github.com/ignitionrobotics/ign-sensors/files/5565124/rgbd_failure.tar.gz)

darksylinc, 2021-06-14:
This suspiciously sounds like https://github.com/ignitionrobotics/ign-rendering/issues/332 which has been fixed now.

Can you check if it has been fixed now with latest ign-rendering5?

> ```
>  Validation Failed: Sampler error:
>   Samplers of different types use the same texture image unit.
>    - or -
>   A sampler's texture unit is out of range (greater than max allowed or negative).
> ```
> 
> That looks suspicious.

Although that sounds suspicious, unfortunately that warning is very common. I never looked deep into this problem, what I think happens is that Ogre assigns 0 to all samplers because our param buffers haven't been initialized yet, calls some GL functions (and _some_ GL drivers warn about this problem at this point), then the params get initialized with correct values and now the shader is actually used.

If the GL driver validates too early, the warning appears. But 99% of the time it's a meaningless warning.