gazebosim/gz-sensors #73: Support grayscale image encodings for cameras

mjcarroll, 2021-01-08:
There are many gray scale cameras on the market.  It would make sense to support this type of image encoding.

Using 8-bit or 16-bit gray scale images would also potentially yield bandwidth savings over having to send full 24 bit RGB images.

adityapande-1995, 2022-05-25:
We could leverage this : https://github.com/gazebosim/gz-rendering/blob/1e415bb1a3d374b15d9c6cb86be76b2b269711fb/include/ignition/rendering/PixelFormat.hh#L37 cc @deepanshubansal01 and make changes here and downstream : https://github.com/gazebosim/gz-sensors/blob/efe31516ea49d43c3be1c093835eba8b6f978c17/src/CameraSensor.cc#L211-L224

iche033, 2022-05-25:
8 bit grayscale support was added recently: https://github.com/gazebosim/gz-sensors/pull/220

deepanshubansal01, 2022-05-26:
> > 8 bit grayscale support was added recently: #220
> 
> Should we add a similar support for 16 bit as well?

Based on the discussion offline, I will proceed with adding 16 bit support as well.

deepanshubansal01, 2022-06-01:
The support for both the 8 bit and 16 bit grayscale image format has been implemented. We can close this issue now ig.

closed 2022-06-01 by deepanshubansal01