gazebosim/gz-sensors #103: List all available sensors on the README

chapulina, 2021-02-27:
A new user landing at https://ignitionrobotics.org/libs/sensors should have a quick way to find out what sensors are supported.

## Desired behavior

List all sensors on the README, with links to their APIs, like [Altimeter](https://ignitionrobotics.org/api/sensors/4.1/classignition_1_1sensors_1_1AltimeterSensor.html).

## Alternatives considered

The list could be somewhere else, but the README is probably the most evident place, both to users navigating to https://ignitionrobotics.org/libs/sensors and to https://github.com/ignitionrobotics/ign-sensors/.

## Implementation suggestion

Use a markdown table or an unordered list.

everly-gif, 2021-02-27:
So I'll Have to add all sensors in the README.md file and when user hovers / clicks on it, it should navigate to the api page. I can do this, Can you assign me this?

benediktkreis, 2022-03-17:
This would be indeed very helpful! Furthermore, I suggest that the README list includes one SDF implementation example and a list of all available tags with a hint if they are required or optional.

tejalbarnwal, 2022-10-08:
Hey, could you please assign this to me?

iche033, 2022-10-10:
> Hey, could you please assign this to me?

done, assigned.

mjcarroll, 2022-10-13:
@tejalbarnwal I think that is a great format.  By available tags, do you mean pointing off to the corresponding `sdformat` documentation?  eg http://sdformat.org/spec?ver=1.9&elem=sensor#sensor_altimeter for altimeter?

tejalbarnwal, 2022-10-13:
I was actually planning to list all the tags individually.
However now I think yeah, this is a better idea to point to the respective `sdformat` documentation.

aagrawal05, 2024-03-30:
Hi, is this issue still active? If so what is remaining to be done?
[The landing page](https://ignitionrobotics.org/libs/sensors) appears to still not have the updated description.
I'd be happy to help.

mpollayil, 2025-01-23:
Any updates on this? As of now it's extremely confusing and hard to use gazebo sensors.

ganatrask, 2025-03-26:
I'd like to work in this issue

kscottz, 2025-03-27:
Go to town! Let us know if you need help.

cc: @iche033

Dakshaarvind, 2025-03-28:
> A new user landing at https://ignitionrobotics.org/libs/sensors should have a quick way to find out what sensors are supported.
>
> ## Desired behavior
> List all sensors on the README, with links to [...]

Hi, I'd like to contribute to making the list of available gazebo sensors for the readme file!

ganatrask, 2025-04-01:
@kscottz Thanks for checking in, I found a bunch of sensors. I made PR #511 for this issue; I believe there would be more sensors supported. How can I find out can you please suggest me something?

iche033, 2025-04-01:
thanks! I've left some comments in your PR.

kscottz, 2025-04-01:
@ganatrask there are most likely additional sensors but @iche033 or @azeey are going to better resources for pointing out that information. My recommendation is to get #511 merged and then circle back with an update.

closed 2025-04-04 by iche033