gazebosim/gz-sensors #107: Add distortion to camera sensors

peci1, 2021-03-10:
## Desired behavior
SDF tags `<distortion>` in camera sensors should be respected.

## Alternatives considered

## Implementation suggestion
As of now, they are silently ignored. I think until distortion is supported, it might be valuable to at least issue a warning that the distortion is not taken into account. It is then also wrong to publish the distortion read from SDF as camera info, as the actual distortion is zero.

## Additional context
This should also appear in the comparison chart between Gazebo Classic and Ignition Gazebo.

chapulina, 2021-03-10:
Here's Gazebo classic's [Camera Distortion tutorial](http://gazebosim.org/tutorials?tut=camera_distortion&cat=sensors) as another reference.

scpeters, 2021-08-12:
much of the implementation of camera distortion in gazebo classic is in the rendering library:

* https://github.com/osrf/gazebo/blob/gazebo11/gazebo/rendering/Distortion.hh

I imagine that when porting to ignition, we will similarly put some of that functionality in ign-rendering?

iche033, 2021-11-16:
We can start by porting Gazebo classic's [Distortion](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/rendering/Distortion.cc) class over to the ogre 1.x implementation in ign-rendering. 

In ign-rendering, we can apply distortion to cameras using a similar approach as applying noise. The idea is to apply distortion as an render pass, see [example](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/examples/render_pass/Main.cc#L156-L166) for applying noise.

We'll need the add the abstract / base Distortion classes, see [GaussianNoisePass.hh](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/include/ignition/rendering/GaussianNoisePass.hh) and [BaseGaussianNoisePass.hh](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/include/ignition/rendering/base/BaseGaussianNoisePass.hh). 

Then put the actual ogre 1.x implementation in `ogre` directory, see [OgreGaussianNoise.cc](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/ogre/src/OgreGaussianNoisePass.cc) and [OgreGaussianNoise.hh](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/ogre/include/ignition/rendering/ogre/OgreGaussianNoisePass.hh)

There is a [macro](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/ogre/src/OgreGaussianNoisePass.cc#L161) to register the render pass so that it can be created by the RenderPassSystem

peci1, 2022-01-31:
https://github.com/ignitionrobotics/ign-rendering/pull/502 has brought the required ign-rendering support.

iche033, 2022-03-01:
https://github.com/ignitionrobotics/ign-sensors/pull/192

WilliamLewww, 2022-03-31:
Closed with the following merges:
* https://github.com/ignitionrobotics/ign-sensors/pull/192
* https://github.com/ignitionrobotics/ign-gazebo/pull/1374

closed 2022-03-31 by WilliamLewww