gazebosim/gz-sensors #111: Default generated thermal camera info topic is incorrect

iche033, 2021-03-22:
## Environment
* OS Version: Ubuntu all version
* Source or binary build?
source and binary

## Description

If no topic is specified in ign-gazebo for a thermal camera, currently the camera_info topic that's generated is missing the thermal camera sensor name, e.g.

```
/world/thermal_camera/model/thermal_camera/link/link/sensor/camera_info
```

The correct topic should be:

```
/world/thermal_camera/model/thermal_camera/link/link/sensor/thermal_camera/camera_info
```

There are a couple ways to fix this. One is shown in https://github.com/ignitionrobotics/ign-sensors/pull/87 which is closed as it changes behavior so it was not merged.

The other method is to make the change on the ign-gazebo side and add a suffix string to thermal camera topic name here:
https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo4/src/rendering/RenderUtil.cc#L1089

i.e. change `const std::string thermalCameraSuffix{""};` to `const std::string thermalCameraSuffix{"image"};`

## Steps to reproduce

1. launch a world with any thermal camera, e.g. the [thermal_camera world](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo4/examples/worlds/thermal_camera.sdf) in ign-gazebo but with [the topic sdf element](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo4/examples/worlds/thermal_camera.sdf#L255) removed.
1. echo the topics by doing `ign topic -l`:
    ```
    /world/thermal_camera/model/thermal_camera/link/link/sensor/camera_info
    /world/thermal_camera/model/thermal_camera/link/link/sensor/thermal_camera
    ```

WagnerMarcos, 2021-05-21:
I have been looking at this and changing `const std::string thermalCameraSuffix{""}; ` to `const std::string thermalCameraSuffix{"/image"};` Is fixing the issue.

I wanted to ask is if this was a Dome version focused Issue, as this is already implemented in Gazebo5 as it can be seen here
https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/src/rendering/RenderUtil.cc#L1121
and in the main branch
https://github.com/ignitionrobotics/ign-gazebo/blob/main/src/rendering/RenderUtil.cc#L1214

iche033, 2021-05-24:
We decided not to target the fix to Dome as it's considered a behavior change so we did the fix to Edifice in https://github.com/ignitionrobotics/ign-gazebo/pull/606

iche033, 2021-05-24:
I think we can close this issue actually since we're likely not going be backporting the fix or finding a different fix for Dome

iche033, 2021-06-07:
fix is in Edifice so closing this issue.

closed 2021-06-07 by iche033