gazebosim/gz-sensors #113: ign-gazebo ECM in Edifice frozen with latest ign-sensors changes

jennuine, 2021-03-26:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
  
  
Source, `main` branch, latest commit 2ce35858ec618542f1253a4d12cc130db94963a3
 
## Description

Running `ign gazebo -r sensors_demo.sdf`

* Expected behavior: Objects falling when starting at a height, camera image streams, and RTF fluctuations when the sim is running

* Actual behavior: Nothing happens

## Steps to reproduce

1. `ign gazebo -r sensors_demo.sdf`

## Output

[image]

After reverting to a commit before this PR https://github.com/ignitionrobotics/ign-sensors/pull/89, it works. Since it seems to be only making `ign-rendering` calls, it could be an `ign-rendering` issue.

chapulina, 2021-03-26:
Let's revert #89 CC @mjcarroll @nkoenig

chapulina, 2021-03-26:
Reverted in #114

closed 2021-03-26 by chapulina