gazebosim/gz-sensors #127: Back faces are culled in sensors (both RGB and depth)

peci1, 2021-04-29:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build? binary 4.1.0
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: ogre2
      - [x] Sensor rendering error. 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on real hardware
    - Rendering system info:

.

    $ LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
    direct rendering: Yes
        Preferred profile: core (0x1)
        Max core profile version: 4.6
        Max compat profile version: 4.6
        Max GLES1 profile version: 1.1
        Max GLES[23] profile version: 3.2
    OpenGL vendor string: AMD
    OpenGL renderer string: AMD RENOIR (DRM 3.40.0, 5.11.15-051115-generic, LLVM 12.0.0)
    OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.0.3 - kisak-mesa PPA
    OpenGL core profile shading language version string: 4.60
    OpenGL core profile context flags: (none)
    OpenGL core profile profile mask: core profile
    OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.0.3 - kisak-mesa PPA
    OpenGL shading language version string: 4.60
    OpenGL context flags: (none)
    OpenGL profile mask: compatibility profile
    OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.0.3 - kisak-mesa PPA
    OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20  
    $           ps aux | grep Xorg
    peci1     2932  2.9  0.3 2267408 113700 tty2   Sl+  dub27  92:55 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
    peci1    16226  0.0  0.0  15672  1024 pts/1    S+   22:53   0:00 grep --color=auto Xorg
    $           sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
    X.Org X Server 1.20.8
    X Protocol Version 11, Revision 0 
    Build Operating System: Linux 4.15.0-140-generic x86_64 Ubuntu
    Current Operating System: Linux cras-17 5.11.15-051115-generic #202104161034 SMP Fri Apr 16 10:40:30 UTC 2021   x86_64
    Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.11.15-051115-generic root=UUID=d982512a-7987-4852-992c-9aeeddc50c46 ro amd_iommu=on
    Build Date: 08 April 2021  01:40:27PM
    xorg-server-hwe-18.04 2:1.20.8-2ubuntu2.2~18.04.5 (For technical support please see http://www.ubuntu.com/support) 
    Current version of pixman: 0.34.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.

## Description
* Expected behavior: Camera sensors do not cull faces (as opposed to GUI cameras)
* Actual behavior: Camera sensors do cull faces.

Some illustration images recorded by sensor cameras:

[image]

[image]

I'm not sure if there are cases when culling would be desirable in sensor cameras... But the images posted above (from tunnel_circuit_practice_02 SubT world) are definitely wrong.

iche033, 2021-04-30:
In this particular case, I think this is a problem mainly with the the world design. It looks like the robot sensor ended up in a place that it shouldn't be and we should fix that. 

We made the exterior part / shell of these tunnel meshes using single sided triangles and oriented with normal facing the inside on purpose so that we can see through them. Disabling culling of these triangles is possible using the [<double_sided>](http://sdformat.org/spec?ver=1.7&elem=material#material_double_sided) param (which disables culling while also making sure lighting is correctly applied) but it's done on a per visual material basis (as opposed to per camera setting). It's off by default mainly as we don't want to render more triangles than necessary. If we enable this for all meshes in the sensor camera view, I think there'll probably be some performance hit.

peci1, 2021-05-01:
Thanks for the explanation. The design of the world is okay, I teleported the robot outside the walls.

Would adding `<force_double_sided>` to cameras make sense?

closed 2021-05-01 by peci1

iche033, 2021-05-05:
> Would adding <force_double_sided> option to cameras make sense?

When we enable double sided triangles on a visual. All cameras will see them as double sided. To selectively enable dthis for one particular camera, we would need to toggle on and off double sided rendering before and after that particular camera update. This is possible but just more performance overhead needed for invalidating and updating the materials between updates.