gazebosim/gz-sensors #131: Clamping for gpu_lidar with noise is causing out of range data to show in ROS2 with ros_ign_bridge.

patrick-neal, 2021-05-16:
I believe the clamping in #86 is causing no return rays to display in Rviz2.
Example of code:
https://github.com/ignitionrobotics/ign-sensors/blob/bb16d12fde391e2e9655c7e74bff85193c00f119/src/Lidar.cc#L202-L220
Comment from existing code base:
https://github.com/ignitionrobotics/ign-sensors/blob/bb16d12fde391e2e9655c7e74bff85193c00f119/src/GpuLidarSensor.cc#L186-L187
Visualization with noise
[image]
Without noise.
[image]

peci1, 2021-06-03:
This has hit us hard in our SubT solution with MARV robot!

[image]

STR:

1. LANG=C ign gazebo -v4 sensors_demo.sdf
2. Insert model https://app.ignitionrobotics.org/OpenRobotics/fuel/models/CTU_CRAS_NORLAB_MARV_SENSOR_CONFIG_1
3. `roscore`
4. `rosrun ros_ign_bridge parameter_bridge /world/lidar_sensor/model/ctu_cras_norlab_marv_sensor_config_1/link/laser/sensor/laser/scan/points@sensor_msgs/PointCloud2[ignition.msgs.PointCloudPacked`
5. open rviz and display the pointcloud

peci1, 2021-06-03:
Fix in #132.

PR #132 (2021-06-04): Remove clamping from lidar noise
# 🦟 Bug fix

Fixes #131

## Summary
This PR removes clamping from application of lidar noise. The clamping generates wrong points for missed rays, as it clamps even the +-Inf values to the lidar range. Removing the clamping might lead to having some points with distance larger than `range_max` of the sensor, but I'd argue that's okay in case of noise. If not, then the code should explicitly check to not clamp infinite values.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (a [... cut]
closed 2021-06-16 by chapulina