gazebosim/gz-sensors #174: Exposing ``SetOrientationReference`` for IMU from sdf

adityapande-1995, 2021-12-08:
## Desired behavior
According to REP 145, an IMU can report orientation with respect to the world, or some given reference frame. The method ``SetOrientationReference`` (https://github.com/ignitionrobotics/ign-sensors/blob/26004b372497ee77616fd23e9abe1d772d79106c/src/ImuSensor.cc#L278) enables this, but has not been exposed vai sdf. There is no way to set this externally.

The test cases verify its usage, but they set it explicitly using ``sensor->SetOrientationReference()`` and ``sensor->SetOrientationEnabled()`` 
(https://github.com/ignitionrobotics/ign-sensors/blob/26004b372497ee77616fd23e9abe1d772d79106c/test/integration/imu.cc#L99-L164) , (https://github.com/ignitionrobotics/ign-sensors/blob/26004b372497ee77616fd23e9abe1d772d79106c/src/ImuSensor_TEST.cc#L316-L370)

It would be nice to have sdf tags (similar to ``<pose>``) to set the reference orientation, and a flag enabling it. The ``pose`` tag is only useful for setting initial orientation, not the reference one.

## Alternatives considered
@scpeters  @chapulina  Open to suggestions !

## Implementation suggestion
My initial thought was to include a custom tag, similar to ``<pose>``, say ``<imu:reference_orintation>`` and parse it in the ``Load`` method for IMU.

scpeters, 2021-12-08:
another option is to leverage `SphericalCoordinates` so that the IMU will specify orientation relative to an ENU, or NED or some other type of frame

adityapande-1995, 2021-12-09:
As per discussion with @scpeters and @chapulina, the ``<orientation_reference_tag>`` specified here : http://sdformat.org/spec?ver=1.9&elem=sensor#imu_orientation_reference_frame will be exposed via ImuSensor.

closed 2021-12-21 by adityapande-1995