gazebosim/gz-sensors #176: Lidar Plugin producing unintended sparse pointcloud

carlos-m159, 2021-12-13:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build?
 Source:  
    branch: ign-sensor6
    commit: 9c075d06b3fc02795d6e24aab9e110f47c6b66b9

Lidar Class does not account  for range equal to **inf**, producing a sparse pointcloud, which by the implementation should not be the intention. 
https://github.com/ignitionrobotics/ign-sensors/blob/9c075d06b3fc02795d6e24aab9e110f47c6b66b9/src/Lidar.cc#L262 

After double checking ign-math:  (branch: **ign-math6**, commit: d338701cd27c172aa931634335024f4c68c78e2c) there is a fixnan function which treats inf and nan as the same, but **isnan** function only verifies if the value is equal to nan. 

Should  **isnan** be refactored? Or do you believe a new guard should be added, like:
 `range = ( ignition::math::isnan(range) ||  std::isinf(range) ) ? ... `  
 

## Description
* Expected behavior: Receiving a Dense Pointcloud. If the echo is NAN or INF, max range should be set.
* Actual behavior: Sparse Pointcloud is received (inf values in the range channel).

iche033, 2021-12-13:
with the proposed changes: `range = ( ignition::math::isnan(range) || std::isinf(range) ) ? ...`, my understanding is that you would like any `+inf` values be returned as `RangeMax()`? 

The lidar return values are implemented based on [ROS REP 117](https://www.ros.org/reps/rep-0117.html) which states:
```
... out of range detections will be represented by +Inf
```

carlos-m159, 2021-12-13:
I was not aware of this. The fact that I raised the issue was mainly due to the fact that I was receiving a pointcloud containing NaNs. Maybe could be related with the bridge when converting the data. I will keep on debugging my test. Thank you for the reference.

closed 2021-12-13 by carlos-m159

carlos-m159, 2021-12-14:
@iche033  So I have referenced the wrong file:
https://github.com/ignitionrobotics/ign-sensors/blob/9c075d06b3fc02795d6e24aab9e110f47c6b66b9/src/GpuLidarSensor.cc#L259

Rephrasing the question after reading REP 117, is correct to label a pointcloud as **dense** but still include non finite (+-Inf) elements inside of it?

carlos-m159, 2021-12-14:
I created a new issue which relates to my second point, since this one was not really describing the problem which we are addressing. https://github.com/ignitionrobotics/ign-sensors/issues/179