gazebosim/gz-sensors #185: Triggered cameras (take photos instead of videos)

chapulina, 2022-01-10:
Gazebo classic supports cameras that only capture images when triggered through ROS, see [gazebo_ros_camera](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/include/gazebo_plugins/gazebo_ros_camera.hpp). When porting that to Ignition, we should make all cameras "triggerable" via an Ignition topic that can be bridged to a ROS topic with `ros_ign`.

## Desired behavior

Cameras that have `<triggered>true</triggered>` will only capture an image when they receive a command on an Ignition topic (like taking a photo instead of recording a video).

## Alternatives considered

Users may be able to achieve this with a custom plugin, but adding this functionality to all cameras by defaults should be easier to maintain and extend. We may also consider adding this functionality to all sensors? I can see this being useful for performance when using lidars, contact sensors, etc.

## Implementation suggestion

1. Add a `<triggered>` parameter to `<sensor>` on SDF. Or just to `<camera>`.
2. Parse that tag in `ign-sensors` and provide the necessary topic and callbacks that enable triggering

mjcarroll, 2022-01-10:
It would also be beneficial to support the `PolledCamera` interface from ROS: https://github.com/ros-perception/image_common/tree/noetic-devel/polled_camera

chapulina, 2022-01-10:
> It would also be beneficial to support the PolledCamera interface from ROS

Nice! I think that could be added to `ros_ign_image`.

closed 2022-05-26 by chapulina