gazebosim/gz-sensors #197: Memory Leak in GpuLidarSensor.cc

paszoujp, 2022-02-24:
## Environment
* OS Version:  Ubuntu 20.04 
* Source or binary build?
  
  
source build: ignition-sensors7 VERSION 7.0.0
branch: main
last commit: 83855aa542275a3b135c2443afeff15a0689ccb3

* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
Graphics: Mesa Intel® UHD Graphics (CML GT2)
CPU: Intel® Core™ i7-10510U CPU @ 1.80GHz × 8 

    - Rendering plugin: [ogre2]. 
      - [Maybe] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [Y] running on real hardware
      - [N] running in virtual machine
      - [N ] running in Docker/Singularity
      - [ N] running remotely (e.g. via SSH)
      - [ N] running in a cloud
      - [ N] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ N] GPU is concurrently used for other tasks
        - [ N] desktop acceleration
        - [ N] video decoding (i.e. a playing Youtube video)
        - [ N] video encoding
        - [ N] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ N] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
```bash
      LANG=C lspci -nn | grep VGA  # might require installing pciutils
       echo "$DISPLAY"
      LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
 ps aux | grep Xorg
sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
    
    - [Y ] Please, attach the ogre.log or ogre2.log file 
    
<details>
  
```bash

jinpei@jinpei-thinkpad:~/workspace/src/ign-sensors$ LANG=C lspci -nn | grep VGA  # might require installing pciutils
00:02.0 VGA compatible controller [0300]: Intel Corporation UHD Graphics [8086:9b41] (rev 02)
jinpei@jinpei-thinkpad:~/workspace/src/ign-sensors$        echo "$DISPLAY"
:0
jinpei@jinpei-thinkpad:~/workspace/src/ign-sensors$       LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) UHD Graphics (CML GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.0.3
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.0.3
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
[... 453 lines cut ...]
        <max_linear_acceleration>1</max_linear_acceleration>
        <min_linear_acceleration>-1</min_linear_acceleration>
        <max_angular_acceleration>2</max_angular_acceleration>
        <min_angular_acceleration>-2</min_angular_acceleration>
        <max_linear_velocity>0.5</max_linear_velocity>
        <min_linear_velocity>-0.5</min_linear_velocity>
        <max_angular_velocity>1</max_angular_velocity>
        <min_angular_velocity>-1</min_angular_velocity>
      </plugin>

    </model>
</sdf>
```

2. use `htop` to monitor the memory usage
3. here's what I got for a tens of few minutes
[image]

[image]

[image]

## Output

[ogre2.log](https://github.com/ignitionrobotics/ign-sensors/files/8130693/ogre2.log)

paszoujp, 2022-02-24:
Adds up.  
I used the following plugin in my world sdf:
```xml
    <plugin
      filename="ignition-gazebo-physics-system"
      name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin
      filename="ignition-gazebo-user-commands-system"
      name="ignition::gazebo::systems::UserCommands">
    </plugin>
    <plugin
      filename="ignition-gazebo-scene-broadcaster-system"
      name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>
    <plugin
      filename="ignition-gazebo-contact-system"
      name="ignition::gazebo::systems::Contact">
    </plugin>
    <plugin
      filename="ignition-gazebo-sensors-system"
      name="ignition::gazebo::systems::Sensors">
      <render_engine>ogre2</render_engine>
    </plugin>
```

andreabernard, 2022-12-29:
Hi all, any news about this  ??

iche033, 2023-01-03:
I just tested this and verified that the still exists in Garden (but not Fortress).

andreabernard, 2023-01-04:
I compare the last commit of the branch  gz-sensors7 with the last commit of the branch ign-sensors6 and I didn't find any changes correlated to this issue

iche033, 2023-01-04:
Found that the memory leak is caused in gz-rendering. 

@darksylinc if you could provide some quick comment and shed some light on the following piece of code, I'll then see if I can create a fix. I found that the vao manager continues to [create const buffers](https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/Ogre2GzHlmsSharedPrivate.cc#L75) because [this condition](https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/Ogre2GzHlmsSharedPrivate.cc#L64-L65) is always true. The pointers (`currPerObjectDataBuffer` and `lastMainConstBuffer`) and being reset in [UnmapObjectDataBuffer](https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/Ogre2GzHlmsSharedPrivate.cc#L108). Perhaps that if  condition for determining when const buffers need to be created has to be changed?

darksylinc, 2023-01-04:
Yep, the logic is faulty.

For reference, we are trying to mimic what [HlmsBufferManager::mapNextConstBuffer](https://github.com/OGRECave/ogre-next/blob/e2287964648740c705da60fd983b068eb4f4f827/Components/Hlms/Common/src/OgreHlmsBufferManager.cpp#L115) does.

I believe this should be the correct code:

```cpp
  if (!this->currPerObjectDataBuffer ||
      this->lastMainConstBuffer != _constBuffers[_currConstBufferIdx])
  {
    // mConstBuffers[this->mCurrentConstBuffer] changed, which means
    // gl_InstanceId / drawId will be reset to 0. We must create a new
    // buffer and bind that one

    UnmapObjectDataBuffer();

    if (_currConstBufferIdx >= this->perObjectDataBuffers.size())
    {
      const size_t bufferSize =
        std::min<size_t>(65536, _vaoManager->getConstBufferMaxSize());
      Ogre::ConstBufferPacked *constBuffer = _vaoManager->createConstBuffer(
        bufferSize, Ogre::BT_DYNAMIC_PERSISTENT, nullptr, false);
      this->perObjectDataBuffers.push_back(constBuffer);
    }

    Ogre::ConstBufferPacked *constBuffer =
      this->perObjectDataBuffers[_currConstBufferIdx];

    this->currPerObjectDataBuffer = constBuffer;
    this->currPerObjectDataPtr = reinterpret_cast<float *>(
      constBuffer->map(0u, constBuffer->getNumElements()));

    GZ_ASSERT(
      _currConstBufferIdx <= _constBuffers.size() &&
        _startMappedConstBuffer != nullptr,
      "This should not happen. Base class must've bound something");

    this->lastMainConstBuffer = _constBuffers[_currConstBufferIdx];

    BindObjectDataBuffer(_commandBuffer, _perObjectDataBufferSlot);
  }
```

All I did was add a `if (_currConstBufferIdx >= this->perObjectDataBuffers.size())` check and then later `Ogre::ConstBufferPacked *constBuffer = this->perObjectDataBuffers[_currConstBufferIdx];`

darksylinc, 2023-01-04:
**Clarification:** `this->perObjectDataBuffers` is supposed to grow up to a certain point (how much depends on how many state swaps happen during render each frame) but at some point it should stabilize and afterwards more entries to `this->perObjectDataBuffers` are only rare or never happen again.

But with the current logic, it's growing indefinitely and without reuse after the frame is done, which is wrong.

Btw I think I'm seeing that we are never destroying those buffers on shutdown. It's not a big deal because OgreNext should deinitialize them for you; but it's for correctness' sake.

iche033, 2023-01-04:
awesome thanks that worked! Memory leak is fixed for me with the changes in https://github.com/gazebosim/gz-rendering/pull/799. I also destroyed the const buffers on exit for correctness' sake

closed 2023-01-06 by iche033