gazebosim/gz-sensors #201: Port "Conform to ros format for header field frame_id of sensor msgs #195" to all ignition versions

halodluc, 2022-03-03:
See #195 for the description. This pull request was merged in branch ign-sensors3, but I think it should be merged probably to all ignition versions. 
I can confirm that it is necessary for ign-sensors6 and I created a new pull request solving my issue #200 
However, I am not sure where else it is necessary and how to do it properly. So this is the reason why I created this issue. 

Environment
* OS Version: Ubuntu 20.04
* Binary installed ignition-edifice with libraries replaced from source ign-sensors6 from #200

chapulina, 2022-03-04:
Hi @halodluc , thank you for opening this issue and PR #200. As @ahcorde mentioned, we prefer to do forward-ports as described on the [contribution guide](https://ignitionrobotics.org/docs/all/contributing#process). I've started with #202 .

halodluc, 2022-03-04:
Hi, thank you very much!

ahcorde, 2022-03-07:
5 :arrow_right: 6 PR https://github.com/ignitionrobotics/ign-sensors/pull/203

doisyg, 2022-04-04:
Already reported there https://github.com/ignitionrobotics/ign-sensors/pull/195, maybe I should open a separate issue but it seems that it works for laserscan msgs but not for pointclouds. I believe the type ignition.msgs.PointCloudPacked is missing the frame field.