gazebosim/gz-sensors #255: Ogre2 / OgreNext not detect when compiling from source

darksylinc, 2022-08-06:
Pinging @mjcarroll 

## Environment
* OS Version:  Ubuntu 20.04
* Source or binary build?
  Bleeding edge main

## Description
* Expected behavior: When dropping Ogre2 into the source code folders and building everything from source (via colcon), Ogre2 is detected.
* Actual behavior: All packages have no issues except gz-sensors which logs:
```
[Processing: gz-gui7, gz-physics6, gz-sensors7]                                                                                        
--- stderr: gz-sensors7                                                                                                                
CMake Warning at /home/matias/Projects/ign/install/lib/cmake/gz-rendering7/gz-rendering7-config.cmake:196 (find_package):
  By not providing "Findgz-rendering7-ogre2.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gz-rendering7-ogre2", but CMake did not find one.

  Could not find a package configuration file provided by
  "gz-rendering7-ogre2" (requested version 7.0.0) with any of the following
  names:

    gz-rendering7-ogre2Config.cmake
    gz-rendering7-ogre2-config.cmake

  Add the installation prefix of "gz-rendering7-ogre2" to CMAKE_PREFIX_PATH
  or set "gz-rendering7-ogre2_DIR" to a directory containing one of the above
  files.  If "gz-rendering7-ogre2" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  /home/matias/Projects/ign/install/share/cmake/gz-cmake3/cmake3/GzUtils.cmake:231 (find_package)
  CMakeLists.txt:72 (gz_find_package)

---
```

Full output:

```
colcon build --cmake-args -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Debug -DBUILD_DOCS=OFF --merge-install
Starting >>> OGRE    
Finished <<< OGRE [1min 15s]
Starting >>> gz-cmake3      
Finished <<< gz-cmake3 [0.72s]                         
Starting >>> gz-utils2
Starting >>> gz-tools2
Finished <<< gz-utils2 [2.46s]                                                                     
Starting >>> gz-math7
Finished <<< gz-tools2 [2.80s]                                                                
Finished <<< gz-math7 [22.4s]                             
Starting >>> gz-common5
Starting >>> gz-msgs9
Starting >>> sdformat13
--- stderr: gz-common5                                                                                                             
Warning, assimp below 5.2.0 detected, setting compatibility mode
---
Finished <<< gz-common5 [20.5s]
Starting >>> gz-plugin2
Finished <<< gz-plugin2 [20.3s]                                                                                                     
Starting >>> gz-rendering7
Finished <<< gz-msgs9 [46.1s]                                                                                                         
Starting >>> gz-transport12
Starting >>> gz-fuel_tools8
Finished <<< gz-fuel_tools8 [16.3s]                                                                                                                                                  
--- stderr: sdformat13                                                                                                                           
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_DOCS

---
Finished <<< sdformat13 [1min 5s]
Starting >>> gz-physics6
Finished <<< gz-transport12 [36.6s]                                                                                                           
--- stderr: gz-rendering7                                                                                   
CUDA_TOOLKIT_ROOT_DIR not found or specified
CMake Warning at /home/matias/Projects/ign/install/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:69 (message):
   CONFIGURATION WARNINGS:
   -- Skipping component [optix]: Missing dependency [OptiX].
      ^~~~~ Set SKIP_optix=true in cmake to suppress this warning.
   
Call Stack (most recent call first):
  CMakeLists.txt:176 (gz_configure_build)

---
Finished <<< gz-rendering7 [53.4s]
Starting >>> gz-gui7
Starting >>> gz-sensors7
[Processing: gz-gui7, gz-physics6, gz-sensors7]                                                                                        
--- stderr: gz-sensors7                                                                                                                
CMake Warning at /home/matias/Projects/ign/install/lib/cmake/gz-rendering7/gz-rendering7-config.cmake:196 (find_package):
  By not providing "Findgz-rendering7-ogre2.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gz-rendering7-ogre2", but CMake did not find one.

  Could not find a package configuration file provided by
  "gz-rendering7-ogre2" (requested version 7.0.0) with any of the following
  names:

    gz-rendering7-ogre2Config.cmake
    gz-rendering7-ogre2-config.cmake

  Add the installation prefix of "gz-rendering7-ogre2" to CMAKE_PREFIX_PATH
  or set "gz-rendering7-ogre2_DIR" to a directory containing one of the above
  files.  If "gz-rendering7-ogre2" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  /home/matias/Projects/ign/install/share/cmake/gz-cmake3/cmake3/GzUtils.cmake:231 (find_package)
  CMakeLists.txt:72 (gz_find_package)

---
Finished <<< gz-sensors7 [30.5s]
Finished <<< gz-gui7 [30.5s]                                                                       
Finished <<< gz-physics6 [1min 20s]                               
Starting >>> gz-sim7
[Processing: gz-sim7]                                    
[Processing: gz-sim7]                                       
[Processing: gz-sim7]                                         
Finished <<< gz-sim7 [1min 38s]                                
Starting >>> gz-launch6
--- stderr: gz-launch6                                       
In file included from /home/matias/Projects/ign/src/ign-launch/src/Manager.cc:62:
/home/matias/Projects/ign/src/ign-launch/src/vendor/backward.hpp: In member function ‘void backward::TraceResolverLinuxImpl<backward::trace_resolver_tag::libdw>::inliners_search_cb::operator()(Dwarf_Die*)’:
/home/matias/Projects/ign/src/ign-launch/src/vendor/backward.hpp:1657:14: warning: switch missing default case [-Wswitch-default]
 1657 |       switch (dwarf_tag(die)) {
      |              ^
/home/matias/Projects/ign/src/ign-launch/src/vendor/backward.hpp: In static member function ‘static Dwarf_Die* backward::TraceResolverLinuxImpl<backward::trace_resolver_tag::libdw>::find_fundie_by_pc(Dwarf_Die*, Dwarf_Addr, Dwarf_Die*)’:
/home/matias/Projects/ign/src/ign-launch/src/vendor/backward.hpp:1729:14: warning: switch missing default case [-Wswitch-default]
 1729 |       switch (dwarf_tag(die)) {
      |              ^
---
Finished <<< gz-launch6 [11.2s]

Summary: 17 packages finished [5min 55s]
  6 packages had stderr output: OGRE gz-common5 gz-launch6 gz-rendering7 gz-sensors7 sdformat13
```

## Steps to reproduce

1. Uninstall deb packages with ogre2
1. Place Ogre2/OgreNext in the same src folder
2. Modify Ogre's CMakeLists.txt so that OGRE_USE_NEW_PROJECT_NAME defaults to TRUE
3. Run colcon

darksylinc, 2022-08-06:
Ahh nevermind.

It's broken for ALL packages, not just gz-sensors. My bad.

closed 2022-08-06 by darksylinc

mjcarroll, 2022-08-12:
This is still broken, letting you know I'm working on it right now.