gazebosim/gz-sensors #284: Garden: Tests for rendering sensors failing on Windows

scpeters, 2022-11-10:
## Environment
* OS Version: Windows CI machine (beefy)
* Source or binary build? source of gz-sensors7

## Description
* Expected behavior: tests that aren't disabled should pass
* Actual behavior: 14 tests are failing https://build.osrfoundation.org/view/ign-garden/job/ign_sensors-gz-7-win/13/testReport/

## Steps to reproduce

1. Run https://build.osrfoundation.org/view/ign-garden/job/ign_sensors-gz-7-win/ build

## Output

From https://build.osrfoundation.org/view/ign-garden/job/ign_sensors-gz-7-win/13/consoleText

~~~
The following tests FAILED:
	 17 - INTEGRATION_camera (Failed)
	 19 - INTEGRATION_depth_camera (Failed)
	 21 - INTEGRATION_distortion_camera (Failed)
	 23 - INTEGRATION_gpu_lidar_sensor (Failed)
	 25 - INTEGRATION_rgbd_camera (Failed)
	 29 - INTEGRATION_thermal_camera (Failed)
	 31 - INTEGRATION_triggered_camera (Failed)
	 33 - INTEGRATION_wide_angle_camera (Failed)
~~~

A representative sample of the console output containing `Ogre/ShadowExtrudePointLight not found`:

~~~
31: [ RUN      ] CameraSensor/TriggeredCameraTest.ImagesWithBuiltinSDF/ogre
31: No HOME defined in the environment. Will not log.
31: [32m[Msg] [m[32mLoading plugin [[m[32mgz-rendering-ogre[m[32m][m[32m
31: [m[36m[Dbg] [C:\Jenkins\workspace\ign_sensors-gz-7-win\ws\gz-sensors\src\CameraSensor.cc:296] [m[36mCamera images for [[m[36mcamera1[m[36m] advertised on [[m[36m/test/integration/TriggeredCameraPlugin_imagesWithBuiltinSDF[m[36m][m[36m
31: [m[36m[Dbg] [C:\Jenkins\workspace\ign_sensors-gz-7-win\ws\gz-sensors\src\CameraSensor.cc:320] [m[36mCamera trigger messages for [[m[36mcamera1[m[36m] subscribed[m[36m on [[m[36m/test/integration/TriggeredCameraPlugin_imagesWithBuiltinSDF/trigger[m[36m][m[36m
31: [m[36m[Dbg] [C:\Jenkins\workspace\ign_sensors-gz-7-win\ws\gz-sensors\src\CameraSensor.cc:601] [m[36mCamera info for [[m[36mcamera1[m[36m] advertised on [[m[36m/test/integration/camera_info[m[36m][m[36m
31: [m[36m[Dbg] [C:\Jenkins\workspace\ign_sensors-gz-7-win\ws\gz-sensors\src\CameraSensor.cc:414] [m[36mEnabling camera sensor: '[m[36mcamera1[m[36m' data [m[36mgeneration.[m[36m
31: [munknown file: error: C++ exception with description "Ogre::RuntimeAssertionException::RuntimeAssertionException: Ogre/ShadowExtrudePointLight not found. Verify that you referenced the 'ShadowVolume' folder in your resources.cfg in Ogre::ShadowVolumeExtrudeProgram::initialise at C:\vcpkg\buildtrees\ogre\src\eddf310f0b-6ab1152694.clean\OgreMain\src\OgreShadowVolumeExtrudeProgram.cpp (line 71)" thrown in the test body.
31: [  FAILED  ] CameraSensor/TriggeredCameraTest.ImagesWithBuiltinSDF/ogre, where GetParam() = "ogre" (1779 ms)

~~~

iche033, 2022-11-11:
Errors from CI:

```
21: [munknown file: error: C++ exception with description "Ogre::RuntimeAssertionException::RuntimeAssertionException: Ogre/ShadowExtrudePointLight not found. Verify that you referenced the 'ShadowVolume' folder in your resources.cfg in Ogre::ShadowVolumeExtrudeProgram::initialise at C:\vcpkg\buildtrees\ogre\src\eddf310f0b-6ab1152694.clean\OgreMain\src\OgreShadowVolumeExtrudeProgram.cpp (line 71)" thrown in the test body.
```

Related issue: https://github.com/gazebosim/gazebo-classic/issues/2700 that suggests something to do with ogre 1.12.x version from conda-forge

azeey, 2022-11-11:
> Related issue: https://github.com/gazebosim/gazebo-classic/issues/2700 that suggests something to do with ogre 1.12.x version from conda-forge

We currently don't use conda in our CI though.

traversaro, 2022-11-15:
For the ShadowVolume issue, i think we just need to add the correct folder to the right env variable (OGRE_RESOURCE_PATH?)  as done for Gazebo Classic at the start of https://github.com/gazebosim/gazebo-classic/issues/2700#issuecomment-949706595 .

azeey, 2022-12-12:
Let's disable these tests for now.