gazebosim/gz-sensors #288: Camera sensor intrinsic parameters `fx` and `fy` are ignored

Myzhar, 2022-11-17:
## Environment
* OS Version: Ubuntu 22.04 
* Source or binary build?
  
  
Gazebo Garden from the sources

## Description
* Expected behavior: 

A simulated camera sensors correctly simulate the intrinsic parameters passed by SDF

* Actual behavior: 

The intrinsic parameters passed to a camera sensor by SDF are ignored.
The same happens if the parameters are forced in C++ code.

## Steps to reproduce

1. Add <intrinsics> fields to a camera sensor in SDF
2. Start simulation
3. Grab a chessboard and perform a calibration

## Output

The calibration performed on the images returned by the simulated camera returns wrong intrinsic parameters.

In my case I set `fx = fy = 700` the calibration returned `fx ~ 1100, fy ~ 1100`.

I tried different tests, changing the dimensions of the chessboard and the positions of the camera, but I always obtain  `fx ~ 1100, fy ~ 1100`.

`cx` and `cy` are instead correctly simulated.

iche033, 2022-11-17:
likely not implemented yet. To support this, I think we just need to add the logic somewhere in the [CameraSensor::CreateCamera()](https://github.com/gazebosim/gz-sensors/blob/116f69e2d4d4a19fca38616bbae8c1213d6c2031/src/CameraSensor.cc#L161) function that gets the lens intrinsics values from SDF and call `camera->SetProjectionMatrix`

Myzhar, 2022-11-17:
This seems to be a serious shortcoming.
How can we simulate a camera if we cannot set camera parameters?
Do you now if I can set `fx` and `fy` indirectly by setting the h-fov and v-fov parameters?

imstevenpmwork, 2022-11-29:
Is there any updates regarding this issue?
Thanks!

iche033, 2022-11-29:
I can take this one. We can port the [functionality](https://github.com/gazebosim/gazebo-classic/blob/gazebo11/gazebo/rendering/Camera.cc#L240-L253) over from gazebo-classic

commit 2ec02005 (2022-12-07): Set custom projection matrix based on intrinsics params from SDF (#293)

Closes #288

This PR extends the camera sensor functionality to support custom projection matrix based on input instrinics values. Specifically, it reads the lens <instrinsics> params from SDF, builds the projection matrix from the input values using code ported from Gazebo-classic, and set the camera to use this custom projection matrix.

Expanded the camera intrinsics integration test to compare image output from cameras with different intrinsics values.
closed 2022-12-07 by iche033