gazebosim/gz-sensors #294: Bounding B

rapop, 2022-12-02:
I'm using the bounding box sensor plugin.
Which has a tutorial here : [tutorials/04_boundingbox_camera.md](https://github.com/AmrElsersy/ign-sensors/blob/BoundingBox/tutorials/04_boundingbox_camera.md)

However, the position x,y,z given by the 3d plugin doesnt seems right.
```
<sensor name="3d" type="boundingbox_camera">
                  <topic>boxes_3d</topic>
                  <camera>
                    <box_type>3d</box_type>
                    <horizontal_fov>1.57</horizontal_fov>
                    <image>
                      <width>800</width>
                      <height>600</height>
                    </image>
                    <clip>
                      <near>0.01</near>
                      <far>1000</far>
                    </clip>
                    <save enabled="true">
                      <path>bounding_box_3d_data</path>
                    </save>
                  </camera>
                  <always_on>1</always_on>
                  <update_rate>30</update_rate>
                  <visualize>true</visualize>
                </sensor>
```

This is the data published on the topic by the plugin.
```
box {
[object_detection_visual-4]   center {
[object_detection_visual-4]     x: 3.0000004768371582
[object_detection_visual-4]     y: -0.39995068311691284
[object_detection_visual-4]     z: -6.4999961853027344
[object_detection_visual-4]   }
[object_detection_visual-4]   orientation {
[object_detection_visual-4]     x: -0.49999845027923584
[object_detection_visual-4]     y: 0.49999833106994629
[object_detection_visual-4]     z: 0.50000202655792236
[object_detection_visual-4]     w: 0.50000166893005371
[object_detection_visual-4]   }
[object_detection_visual-4]   boxSize {
[object_detection_visual-4]     x: 1
[object_detection_visual-4]     y: 1
[object_detection_visual-4]     z: 1
[object_detection_visual-4]   }
[object_detection_visual-4] }
[object_detection_visual-4] label: 10
```

But the label 10 in my scene is not at that position.

```
        <model name="box1">
              <pose>4 -3 0.5 0 0 0</pose>
              <link name="box_link">
                <inertial>
                  <inertia>
                    <ixx>1</ixx>
                    <ixy>0</ixy>
                    <ixz>0</ixz>
                    <iyy>1</iyy>
                    <iyz>0</iyz>
                    <izz>1</izz>
                  </inertia>
                  <mass>1.0</mass>
                </inertial>
                <collision name="box_collision">
                  <geometry>
                    <box>
                      <size>1 1 1</size>
                    </box>
                  </geometry>
                </collision>

                <visual name="box_visual">
                  <geometry>
                    <box>
                      <size>1 1 1</size>
                    </box>
                  </geometry>
                  <material>
                    <ambient>0 0 0.5 1</ambient>
                    <diffuse>0 0 1 1</diffuse>
                    <specular>0 0 0.3 1</specular>
                  </material>
                  <plugin filename="gz-sim-label-system" name="gz::sim::systems::Label">
                    <label>10</label>
                  </plugin>
                </visual>
              </link>
        </model>
```

Do I need to do some kind of transformation? 
Has this plugin been tested with the 3d positions given?

Thanks.

[wheeled_robot.zip](https://github.com/gazebosim/gz-sensors/files/10142029/wheeled_robot.zip)

AmrElsersy, 2022-12-03:
As mentioned in the sensor msg's [documentation](https://github.com/gazebosim/gz-msgs/blob/gz-msgs9/proto/gz/msgs/oriented_3d_box.proto) the 3d bounding boxes are in 3d camera coordinates, this is according to some famous datasets in computer vision which give a label in 3d camera coordinates

so you have to transform them back to world coordinates to get the position in the world you want

azeey, 2022-12-05:
Closing since the plugin is behaving as expected. Feel free to reopen if that's not the case.

closed 2022-12-05 by azeey