gazebosim/gz-sensors #306: Using gz_frame_id emits a warning; should be part of the SDFormat spec

azeey, 2023-01-06:
## Environment
* OS Version: any
* Source or binary build?  any
 
## Description
* Expected behavior: No warnings when using `gz_frame_id` to set a custom frame id for a sensor in gz-sim. 
* Actual behavior: The following error is emitted warning that the element `gz_frame_id` is not part of the SDFormat spec.
 
```
Warning [Utils.cc:130] [/sdf/world[@name="imu_sensor"]/model[@name="imu_model"]/link[@name="link"]/sensor[@name="imu_sensor"]/gz_frame_id:/home/addisu/ws/garden/src/gz-sim/test/worlds/imu.sdf:L75]: XML Element[gz_frame_id], child of element[sensor], not defined in SDF. Copying[gz_frame_id] as children of [sensor].
```

I think we'd want this tag in SDFormat. I would also say we could just use `frame_id` instead of `gz_frame_id`.

## Steps to reproduce

1. Modify `gz-sim/test/worlds/imu.sdf` to use `gz_frame_id` 
2. Run the world in gz sim.

azeey, 2023-08-04:
@ahcorde is this something we can do for Harmonic?

muttistefano, 2024-01-05:
Hi all, 
as of today is there a way to change the frame_id of a sensor?
I am using a gpu_lidar but I cannot manage to change the frame_id attached to the message.
Thanks

azeey, 2024-04-04:
> Hi all, as of today is there a way to change the frame_id of a sensor? I am using a gpu_lidar but I cannot manage to change the frame_id attached to the message. Thanks

Yes, you should be able to use `gz_frame_id`. In Fortress, you can use `ign_frame_id` (discouraged) or `gz_frame_id`.

faizurrahman1998, 2024-06-30:
Hello everyone,
I am facing the same problem. `gz_frame_id` tag is not changing the frame_id of the lidar_sensor. Rather, the lidar_sensor is publishing messages with frame_id of the format: `ROBOT_NAME/base_footprint/NAME_OF_THE_SENSOR`. is there any new fix for this?

iche033, 2024-07-03:
included frame id fix for lidar sensor in https://github.com/gazebosim/gz-sensors/pull/444. I'll backport to gz-sensors8 (harmonic) and make it compatible with `<gz_frame_id>` once that PR is merged.

iche033, 2024-07-10:
backported fix in #446

closed 2024-07-10 by iche033

seddiqbenterki, 2025-01-02:
i am using gazebo harmonic with ros2 jazzy  i still face the same problem of gz-frame-id is there any reference code to check is the problem from my side or the bug is still not fixed

TZECHIN6, 2025-01-09:
I am learning to use the new gazebo, but I got very confused about the tag used for the sensor espically. In [here](http://sdformat.org/spec?ver=1.9&elem=sensor#sensor_lidar), there is no tag called `gz_frame_id`, and also not all example showing that all the sensor need to assign a frame, e.g. IMU has no specified frame_id... so what is this actually doing? And so will it break ROS2 to understand the publish frame_id?

iche033, 2025-01-09:
[gz_frame_id](https://github.com/gazebosim/gz-sensors/blob/aaef365e6c51dcf98334e9f7a70e5fc4e3281f91/src/Sensor.cc#L187) is a custom tag (not part of the official sdf spec) that's read by gazebo. It's available in Fortress but deprecated in Ionic. It's replaced by `//sensor/fram_id` (https://github.com/gazebosim/sdformat/pull/1454) in sdf 1.12. If specified, it'll populate the msg header with this frame id value.

TZECHIN6, 2025-01-13:
Thanks @iche033, now from the code you referenced to, I have more  understanding the purpose of the tag, and its corresponding tag name. Any official document I can reference to? 

for example https://gazebosim.org/api/gazebo/6/classignition_1_1gazebo_1_1systems_1_1MecanumDrive.html, this API document clearly show the useable parameter for the plugin mecanum drive.

iche033, 2025-01-13:
I couldn't find the documentation for this. I found the original discussion that led to this tag: https://github.com/gazebosim/ros_gz/pull/111, which was added in https://github.com/gazebosim/gz-sensors/pull/195

PerFrivik, 2025-04-18:
Sorry is this also supposed to work for the imu? When I set the gz_frame_id, it does not work / change the frame id of my imu sensor.

iche033, 2025-04-18:
> Sorry is this also supposed to work for the imu? When I set the gz_frame_id, it does not work / change the frame id of my imu sensor.

it should. I tested this with the [imu sensor](https://github.com/gazebosim/gz-sim/blob/db31641f7a07fb9d179994442d090477d9848396/examples/worlds/sensors.sdf#L171-L177) in the example sensors.sdf world and I see the msg header now contains the frame_id that I set to:

Added `gz_frame_id`:
```diff
diff --git a/examples/worlds/sensors.sdf b/examples/worlds/sensors.sdf
index 9717bb11b..096b850ba 100644
--- a/examples/worlds/sensors.sdf
+++ b/examples/worlds/sensors.sdf
@@ -168,6 +168,7 @@
           </air_speed>
         </sensor>
         <sensor name="imu" type="imu">
+          <gz_frame_id>test_frame_id</gz_frame_id>
           <always_on>1</always_on>
           <update_rate>100</update_rate>
           <visualize>true</visualize>
```

Launch the example world and `gz topic -e -t  /imu | grep -A 10 header`:

```pbtxt
header {
  stamp {
    sec: 48
    nsec: 170000000
  }
  data {
    key: "frame_id"
    value: "test_frame_id"
  }
```

yyagin, 2025-04-21:
I am using ROS 2 Jazzy and Gazebo Harmonic. Neither the `<gz_frame_id>` nor the `<frame_id>` tag works for the IMU and NavSat plugins. I tried using `<frame_id>` based on [gazebosim/sdformat#1454](https://github.com/gazebosim/sdformat/pull/1454), but I believe it is not available in the Harmonic release.

PerFrivik, 2025-04-22:
@iche033 it works now! I think when I ctrl-c'd gazebo there was some instance still alive hahaha, so before killing it in htop it was never updated correctly... thanks!

stevenbrills, 2026-02-19:
@iche033 Harmonic is stuck in a weird place where neither <gz_frame_id>, <ign_frame_id> or <frame_id> is recognized as a legal element. Is there another way to fix this? The sensor name used in place of the frame id discussed [here](https://github.com/gazebosim/ros_gz/pull/111) also doesn't work in Harmonic.

stevenbrills, 2026-02-19:
Yeah as others have highlighted, this is what Harmonic says when I use the tag `<gz_frame_id>`. The image below shows the error. Not just that, I think some other comments here also reference the same issue. None of the frame tags are supported in Harmonic.

[image]