gazebosim/gz-sensors #318: Release of libignition-sensors6

livanov93, 2023-02-13:
I am interesed in release of new `libignition-sensors6`.

What is the release process and what needs to be done in case new binary is needed?
Is there a roadmap/document that sets the expectations/requirements that need to be satisfied to release new package?

iche033, 2023-02-13:
there is no fixed schedule / doc for releases. It's currently done on demand. A pre-release is being made in https://github.com/gazebosim/gz-sensors/pull/320. A stable 6.7.0 release will be made shortly afterwards.

livanov93, 2023-02-14:
@iche033 I can see the CI is not passing on #320 for jammy.

iche033, 2023-02-14:
jammy rendering tests have been failing on github actions for some time. Something that I need to look at

livanov93, 2023-02-15:
@iche033 Does that disables me in getting a [pre-released version](http://packages.osrfoundation.org/gazebo/ubuntu-prerelease/pool/main/i/ignition-sensors6/)?

iche033, 2023-02-15:
The debbuilds failed so the release didn't go out. I created https://github.com/gazebo-release/gz-sensors6-release/pull/15 to fix the issue.

closed 2023-02-17 by livanov93