gazebosim/gz-sensors #325: Noise of navsat position horizontal in degree

vincentrou, 2023-02-24:
## Description
* Expected behavior: Setting noise in meters when using navsat/position_sensing/horizontal/noise/stddev as specified here : http://sdformat.org/spec?ver=1.9&elem=sensor#position_sensing_horizontal
* Actual behavior: Setting noise in degrees : https://github.com/gazebosim/gz-sensors/blob/gz-sensors7/src/NavSatSensor.cc#L178

rachit-aggarwal, 2024-03-26:
I noticed the same issue when trying to setup using Fortress. Wanted to see if there is any update on this?

Not sure, if it helps, I noticed that in the [SDF documentation](http://sdformat.org/spec?ver=1.11&elem=sensor#position_sensing_horizontal), the user is expected to prescribe noise in meters.

Kametor, 2024-12-03:
I am experiencing the same issue. Has there been any progress on this or any workaround in the meantime?
Thanks

peci1, 2025-06-11:
Related issue in https://github.com/gazebosim/sdformat/issues/1572 .

mr3screen, 2026-11-09:
Hi @mjcarroll, where you able to find the original intent?