gazebosim/gz-sensors #328: :farmer: `triggered_boundingbox_camera` integration test failing in gz-sensors7

Crola1702, 2023-02-28:
## Environment
* OS Version: Ubuntu 20.04
* Binary Gazebo Garden

## Description
* Expected behavior: triggered_boundingbox_camera should pass
* Actual behavior: triggered_boundingbox_camera fails

## Steps to reproduce
1. Run a build in [gz-sensors7](https://build.osrfoundation.org/job/ignition_sensors-ci-gz-sensors7-focal-amd64/) job
2. See [BoundingBoxCameraSensor/TriggeredBoundingBoxCameraTest.BoxesWithBuiltinSDF/ogre2](https://build.osrfoundation.org/job/ignition_sensors-ci-gz-sensors7-focal-amd64/32/testReport/(root)/BoundingBoxCameraSensor_TriggeredBoundingBoxCameraTest/BoxesWithBuiltinSDF_ogre2/) test fail

## Output

Reference build: https://build.osrfoundation.org/job/ignition_sensors-ci-gz-sensors7-focal-amd64/32/

Test regression: [BoundingBoxCameraSensor/TriggeredBoundingBoxCameraTest.BoxesWithBuiltinSDF/ogre2](https://build.osrfoundation.org/job/ignition_sensors-ci-gz-sensors7-focal-amd64/32/testReport/(root)/BoundingBoxCameraSensor_TriggeredBoundingBoxCameraTest/BoxesWithBuiltinSDF_ogre2/)

Log output:
```
/home/jenkins/workspace/ignition_sensors-ci-gz-sensors7-focal-amd64/ign-sensors/test/integration/triggered_boundingbox_camera.cc:205
Expected equality of these values:
  g_boxes.size()
    Which is: 0
  size_t(2)
    Which is: 2
```

The test was added here: https://github.com/gazebosim/gz-sensors/pull/322

And log output refers to this line of code: [triggered_boundingbox_camera.cc#L205](https://github.com/gazebosim/gz-sensors/blob/gz-sensors7/test/integration/triggered_boundingbox_camera.cc#L205)

Crola1702, 2023-02-28:
This test behaviour is weird on CI:
 * It passes on drogon ([gz-sensors-build](https://build.osrfoundation.org/job/ignition_sensors-ci-gz-sensors7-focal-amd64/34/) & [pr's CI build](https://build.osrfoundation.org/job/ignition_sensors-ci-pr_any-ubuntu_auto-amd64/1360/))
 * It fails on [optimus](https://build.osrfoundation.org/job/ignition_sensors-ci-gz-sensors7-focal-amd64/33/) and [r2d2](https://build.osrfoundation.org/job/ignition_sensors-ci-gz-sensors7-focal-amd64/32/)

I think these are slower machines. Maybe we could increase the timeout for the last part of the test (despite its 10s right now)

iche033, 2023-02-28:
should be fixed by https://github.com/gazebosim/gz-sensors/pull/329

PR #329 (2023-02-28): Fix flaky  bounding box camera test

# 🦟 Bug fix

Fixes #328

## Summary

Fixes flaky bounding box camera test due to timing / race condition. This PR fixes the test by making sure:
1. we allow time for trigger msg to arrive to bounding box camera
2. make sure msgs are received on the box topic (instead of image topic) which is what the tests is checking
3. add mutex to protect the `isTriggered` variable.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if  [... cut]
closed 2023-02-28 by iche033