gazebosim/gz-sensors #360: Disrepancy between gazebo IMU noise mdoel and Rotors IMU noise model

TanguyTheOtter, 2023-07-17:
Hi! I read in the `GaussianNoiseModel::ApplyImpl(double _in, double _dt)` function that it's implementation is based on [the one available in Rotors](https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo_plugins/src/gazebo_imu_plugin.cpp). By comparing both of them, I notice that a "-1" had been removed in [the gazebo model's calculation of sigma_b_d](https://github.com/gazebosim/gz-sensors/blob/a5a0c348a53b17f963d8fccfc99af4f3ca8a79b8/src/GaussianNoiseModel.cc#L132C1-L132C1) :

- gazebo : `double sigma_b_d = sqrt(-sigma_b * sigma_b * tau / 2 * expm1(-2 * _dt / tau));`
- rotors : `double sigma_b_a_d = sqrt(-sigma_b_a * sigma_b_a * tau_a / 2.0 * (exp(-2.0 * dt / tau_a) - 1.0));`

Is there a particular reason behind this change? I'd be interested to know.

Additionally, the gazebo noise model adds an extra sign randomization step to the bias in :

```
// With equal probability, we pick a negative bias (by convention,
  // rateBiasMean should be positive, though it would work fine if
  // negative).
  if (math::Rand::DblUniform() < 0.5)
    this->dataPtr->bias = -this->dataPtr->bias;
```

Which I found no equivalent for in the rotors code and would be interested in knowing the motivation for.

Lastly, it is said that "More information about the parameters and their derivation can be found at https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model". However, I find no mention of the dynamicBiasStdDev (sigma_b) and dynamicBiasCorrTime (tau) parameters, or how to obtain them. Could you share which IMU calibration method you used to obtain these? I found some values in the files for the [gazebo-fuel X4 drone model](https://app.gazebosim.org/OpenRobotics/fuel/models/X4%20UAV%20Config%204) but they seem quite platform-specific.

For context, I'm trying to match my gazebo simulation to tests I'm running outdoors with an IMU calibrated by Kalibr, to see how my filter performs in sim vs in real life. This is why I'm trying to understand the differences between both implementations, and decide whether to write a gazebo plugin to ensure they match.

Thank you!

mjcarroll, 2023-07-17:
> Is there a particular reason behind this change? I'd be interested to know.

For this change, we make use of [`std::expm1`](https://en.cppreference.com/w/cpp/numeric/math/expm1)

Computes the e ([Euler's number](https://en.wikipedia.org/wiki/E_(mathematical_constant)), 2.7182818...) raised to the given power num, minus 1.0. This function is more accurate than the expression [std::exp](http://en.cppreference.com/w/cpp/numeric/math/exp)(num) - 1.0 if num is close to zero.

> Additionally, the gazebo noise model adds an extra sign randomization step to the bias in :

This predates the inclusion of the rotors noise model, but my assumptions is that because the bias is drawn from a gaussian distribution specified by mean/stddev, we want to make sure that it can additionally be drawn as a positive or negative bias.

In the case that bias mean was 1, bias stddev is 0.1, there is also the chance that the bias may be centered around -1, which is handled by the sign flip.

TanguyTheOtter, 2023-07-17:
> For this change, we make use of [`std::expm1`](https://en.cppreference.com/w/cpp/numeric/math/expm1)
>
> Computes the e ([Euler's number](https://en.wikipedia.org/wiki/E_(mathematical_constant)), [...]

Ah I see that makes alot more sense, thank you for your lighting-fast reply!

mjcarroll, 2023-07-17:
> Lastly, it is said that "More information about the parameters and their derivation can be found at https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model". However, I find no mention of the dynamicBiasStdDev (sigma_b) and dynamicBiasCorrTime (tau) parameters

Those are also copied from rotors: 
https://github.com/ethz-asl/rotors_simulator/blob/513bb92da0c1a0c968bdc679dffc8fe7d77de918/rotors_gazebo_plugins/include/rotors_gazebo_plugins/gazebo_imu_plugin.h#L39-L55

computing tau can be done using the allan variance method. I have a script that does it from gazebo data here: https://github.com/gazebosim/gz-sensors/blob/gz-sensors7/examples/imu_noise/plot_samples.py  and there is another ROS package that does the same thing here: https://github.com/ori-drs/allan_variance_ros  

It is the x (tau) value of the minimum of the allan variance plot, like this one from the kalibr wiki page:
[image]

I believe the reason you don't see it in the kalibr parameters is because while we (and rotors) simulate it, kalibr does not make use of it.

mjcarroll, 2023-07-17:
> Ah I see that makes alot more sense, thank you for your lightning-fast reply!

Happy to help, let me know if you uncover any more oddities or concerns.

TanguyBJG, 2023-07-18:
> > Ah I see that makes alot more sense, thank you for your lightning-fast reply!
> 
> Happy to help, let me know if you uncover any more oddities or concerns.

Thanks! Is dynamicBiasStdDev (sigma_b) in the gazebo IMU model the same as sigma(tau) In the [gazebo script you linked](https://github.com/gazebosim/gz-sensors/blob/gz-sensors7/examples/imu_noise/plot_samples.py)?

mjcarroll, 2023-07-18:
I believe you are looking for this: 

https://github.com/gazebosim/gz-sensors/blob/a5a0c348a53b17f963d8fccfc99af4f3ca8a79b8/examples/imu_noise/plot_samples.py#L67

For that minimum point in the graph.  `tau_b` is the correlation time (X value).  `b` is the bias instability (Y value).

`b` is the value you are looking for.

TanguyTheOtter, 2023-07-18:
> I believe you are looking for this:
>
> https://github.com/gazebosim/gz-sensors/blob/a5a0c348a53b17f963d8fccfc99af4f3ca8a79b8/examples/imu_noise/plot_samples.py#L67
>
> For that minimum point in the [...]

I see, one last question :

- dynamicBiasStdDev = sigma_b = b = bias instability = y-value at the minima of a which curve(s)?
- dynamicBiasCorrTime = tau = tau_b = correlation time = x-value at the minima of which curve(s)?

Thanks you for your responsivity!

mjcarroll, 2023-07-18:
> dynamicBiasStdDev = sigma_b = b = bias instability = y-value at the minima of a which curve(s)?
> dynamicBiasCorrTime = tau = tau_b = correlation time = x-value at the minima of which curve(s)?

Correct.  If you are planning to reproduce this on a real, physical IMU, make sure that you capture a long enough sequence of data (a few hours) to be sure that you can get the full graph.

TanguyTheOtter, 2023-07-18:
> Correct. If you are planning to reproduce this on a real, physical IMU, make sure that you capture a long enough sequence of data (a few hours) to be sure that you can get the full graph.

Yep will run it overnight, could you clarify which curve I should be taking the minima (x=tau, y=b) of? That wasnt super clear to me :)

closed 2023-07-18 by TanguyTheOtter

mjcarroll, 2023-07-18:
> Yep will run it overnight, could you clarify which curve I should be taking the minima (x=tau, y=b) of? That wasnt super clear to me

It comes from the allan variance curve.  You will have one per channel of your IMU (3x for accel, 3x for gyro).  The minimum of each of those curves will be what you are looking for for bias instability.

TanguyTheOtter, 2023-07-19:
> It comes from the allan variance curve. You will have one per channel of your IMU (3x for accel, 3x for gyro). The minimum of each of those curves will be what you are looking for for bias [...]

Thank you for confirming that, have a great day!