gazebosim/gz-sensors #384: optical_frame_id causing error in parsing sdf

ashBabu, 2023-09-14:
ROS2 Humble, Gazebo Fortress. Installed everything through `sudo apt install ...`

I added the following in my urdf where name is `camera`
```
<gazebo reference="${name}_bottom_screw_frame">
            <sensor name="realsense_d435" type="rgbd_camera">
                <camera>
                 <optical_frame_id>"${name}_color_optical_frame"</optical_frame_id>
                    <horizontal_fov>1.047</horizontal_fov>
                    <image>
                        <width>640</width>
                        <height>480</height>
                        <format>RGB_INT8</format>
                    </image>
                    <clip>
                        <near>0.1</near>
                        <far>5</far>
                    </clip>
                    <distortion>
                        <k1>0.0</k1>
                        <k2>0.0</k2>
                        <k3>0.0</k3>
                        <p1>0.0</p1>
                        <p2>0.0</p2>
                        <center>0.5 0.5</center>
                    </distortion>
                    <lens>
                        <intrinsics>
                            <fx>554.25469</fx>
                            <fy>554.25469</fy>
                            <cx>320.5</cx>
                            <cy>240.5</cy>
                            <s>0</s>
                        </intrinsics>
                    </lens>
                    <noise>
                        <type>gaussian</type>
                        <mean>0</mean>
                        <stddev>0.007</stddev>
                    </noise>
                    <depth_camera>
                        <clip>
                            <near>0.1</near>
                            <far>5</far>
                        </clip>
                    </depth_camera>
                </camera>
                <always_on>1</always_on>
                <update_rate>30</update_rate>
                <visualize>true</visualize>
                <topic>${topic_ns}/${topic}</topic>
                <enable_metrics>true</enable_metrics>
                <plugin
                        filename="ignition-gazebo-sensors-system"
                        name="ignition::gazebo::systems::Sensors">
                    <render_engine>ogre2</render_engine>
                </plugin>
            </sensor>
        </gazebo>
```
and the corresponding `ros_gz_bridge` to get the topics available to ROS. This publishes camera topics under a strange frame id  (`robot_name/link_name/sensor_name`) which is not available in the tf tree. Then I added an `<optical_frame_id>` but this causes an error as follows. 
```
[ign-2] [libprotobuf ERROR google/protobuf/text_format.cc:309] Error parsing text-format ignition.msgs.EntityFactory: 1:2109: Message type "ignition.msgs.EntityFactory" has no field named "camera_color_optical_frame".
```

ahcorde, 2023-11-10:
You should use 

```xml
                <ignition_frame_id>"${name}_color_optical_frame"</ignition_frame_id>
```

let me know if you fix your issue with this fix

ashBabu, 2023-11-10:
Thanks @ahcorde. I already fixed this with a slight modification to what you have suggested. What works for me is providing `ignition_frame_id` without `" "`. So it would be ` <ignition_frame_id>${name}_color_optical_frame</ignition_frame_id>`. 
Closing this

closed 2023-11-10 by ashBabu

stanislawix, 2025-01-14:
So in the newer Gazebo Harmonic you should use <optical_frame_id> tag or some other tag, like <gz_frame_id>? Asking beacuse of the (ign, ignition) -> (gz, gazebo) renaming thing happening while migrating from Ignition to Gazebo Harmonic.