gazebosim/gz-sensors #393: :farmer: 4 `INTEGRATION` tests failing in MacOS Monterey and Ventura in gz-sim3

Crola1702, 2023-10-11:
## Environment
* OS Version: MacOS Monterey and Ventura
* Source gz-sim3-homebrew

## Description
* Expected behavior: All tests should pass
* Actual behavior: 4 INTEGRATION tests are failing 

## Steps to reproduce

1. Run a build in [ignition_sensors-ci-ign-sensors3-homebrew-amd64](https://build.osrfoundation.org/job/ignition_sensors-ci-ign-sensors3-homebrew-amd64/)
2. See INTEGRATION tests fail

## Output

Reference build: https://build.osrfoundation.org/job/ignition_sensors-ci-ign-sensors3-homebrew-amd64/111/

Test regressions:
* [INTEGRATION_depth_camera_plugin.test_ran](https://build.osrfoundation.org/job/ignition_sensors-ci-ign-sensors3-homebrew-amd64/111/testReport/junit/(root)/INTEGRATION_depth_camera_plugin/test_ran/)
* [INTEGRATION_gpu_lidar_sensor_plugin.test_ran](https://build.osrfoundation.org/job/ignition_sensors-ci-ign-sensors3-homebrew-amd64/111/testReport/junit/(root)/INTEGRATION_gpu_lidar_sensor_plugin/test_ran/)
* [INTEGRATION_rgbd_camera_plugin.test_ran](https://build.osrfoundation.org/job/ignition_sensors-ci-ign-sensors3-homebrew-amd64/111/testReport/junit/(root)/INTEGRATION_rgbd_camera_plugin/test_ran/)
* [INTEGRATION_thermal_camera_plugin.test_ran](https://build.osrfoundation.org/job/ignition_sensors-ci-ign-sensors3-homebrew-amd64/111/testReport/junit/(root)/INTEGRATION_thermal_camera_plugin/test_ran/)

Log output (almost the same for each of the regressions):
<details>

```
15: [ RUN      ] DepthCameraSensor/DepthCameraSensorTest.ImagesWithBuiltinSDF/ogre2
15: [Err] [Ogre2RenderEngine.cc:432] Unable to find Ogre Plugin[/usr/local/opt/ogre2.1/lib/OGRE-2.1/RenderSystem_GL3Plus]. Rendering will not be possible.Make sure you have installed OGRE properly.
15: 2023-10-11 08:51:31.936 INTEGRATION_depth_camera_plugin[63405:13931372] XType: failed to connect - Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service named com.apple.fonts was invalidated: failed at lookup with error 3 - No such process." UserInfo={NSDebugDescription=The connection to service named com.apple.fonts was invalidated: failed at lookup with error 3 - No such process.}
15: 2023-10-11 08:51:31.936 INTEGRATION_depth_camera_plugin[63405:13931372] Font server protocol version mismatch (expected:5 got:0), falling back to local fonts
15: 2023-10-11 08:51:31.936 INTEGRATION_depth_camera_plugin[63405:13931372] XType: unable to make a connection to the font daemon!
15: 2023-10-11 08:51:31.936 INTEGRATION_depth_camera_plugin[63405:13931372] XType: XTFontStaticRegistry is enabled as fontd is not available.
15/32 Test #15: INTEGRATION_depth_camera_plugin .............***Exception: SegFault  3.13 sec
```

</details>

This is a regression that is happening only on Monterey and Ventura machines (that currently are all of our MacOS machines).

It has failed since the first build in mac-five.monterey ([build 75](https://build.osrfoundation.org/job/ignition_sensors-ci-ign-sensors3-homebrew-amd64/75/)). Now that we have only Monterey and Ventura machines, it's failing consistently.

azeey, 2023-10-16:
Let's check if there's something quick to backport, otherwise, let's disable these test on Citadel.

scpeters, 2023-10-16:
> Let's check if there's something quick to backport, otherwise, let's disable these test on Citadel.

candidate quick backport in https://github.com/gazebosim/gz-sensors/pull/397

azeey, 2023-10-16:
Fixed by #397

PR #397 (2023-10-16): Backport rendering test fixes from fortress to citadel
# 🦟 Bug fix

Fixes #393.

## Summary

Backports #169, and #324 to citadel to fix rendering tests

## Checklist
- [X] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))
- [ ] While waiting fo [... cut]
closed 2023-10-16 by azeey