gazebosim/gz-sensors #415: Turn on bias factor

frede791, 2024-02-22:
## Desired behavior
The IMU plugin in old gazebo used a factor called `turn_on_bias_sigma` factor for modelling noise. 
https://github.com/PX4/PX4-SITL_gazebo-classic/blob/33ac87a37676fb597de110e926bbf0a080169ffe/src/gazebo_imu_plugin.cpp#L173C3-L180C4
As far as I can tell there is currently no way to model this in any way? Would it be possible to add this as a specifiable tag that can be used in the noise model?

## Implementation suggestion
Add an additional tag that can be read out in `GaussianNoiseModel.cc` and added as a factor in the `Load()` function

azeey, 2024-03-12:
Is this different from the [bias_mean](http://sdformat.org/spec?ver=1.11&elem=sensor#noise_bias_mean) and [bias_stddev](http://sdformat.org/spec?ver=1.11&elem=sensor#noise_bias_stddev) parameters?

frede791, 2024-03-13:
Yes, as far as I understand, the noise model that is employed is slightly different to the old one that was used previously.

azeey, 2024-03-14:
Could you elaborate more? Based on the lines of code you linked to, it seems to me that we can set `bias_mean=0` and `bias_stddev=1.0` to get the same results. cc @mjcarroll

mjcarroll, 2024-03-14:
> Based on the lines of code you linked to, it seems to me that we can set bias_mean=0 and bias_stddev=1.0 to get the same results

Yes, this should accomplish the same results.  With the IMU, we model both a static (turn-on) bias as well as a dynamic (time varying) bias.

The turn on bias is drawn from a gaussian distribution modelled with mean `bias_mean` and stddev `bias_stddev` when the sensor is loaded.