gazebosim/gz-sensors #423: Published CameraInfo does not contain proper information

wittenator, 2024-04-04:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
  Binary: Harmonic

## Description
* Expected behavior: The camera info topic should correctly reflect the intrinsic properties of the camera
* Actual behavior: The camera info contains default values that do no get changed.

## Steps to reproduce and output
1. Insert a camera sensor with non-default parameters e.g.
```
<sensor name="camera" type="camera">
          <gz_frame_id>camera_link_optical</gz_frame_id>
          <camera>
            <horizontal_fov>1.478294</horizontal_fov>
            <vertical_fov>1.153663</vertical_fov>
            <image>
              <width>640</width>
              <height>480</height>
            </image>
            <clip>
              <near>0.01</near>
              <far>60</far>
            </clip>
          </camera>
          <always_on>1</always_on>
          <update_rate>20</update_rate>
          <visualize>false</visualize>
        </sensor>
```
3. Subscribe to the /camera_info topic. This yields the following message:
```
header {
  stamp {
    sec: 47
  }
  data {
    key: "frame_id"
    value: "camera_link_optical"
  }
}
width: 640
height: 480
distortion {
  k: 0
  k: 0
  k: 0
  k: 0
  k: 0
}
intrinsics {
  k: 277
  k: 0
  k: 160
  k: 0
  k: 277
  k: 120
  k: 0
  k: 0
  k: 1
}
projection {
  p: 277
  p: 0
  p: 160
  p: 0
  p: 0
  p: 277
  p: 120
  p: 0
  p: 0
  p: 0
  p: 1
  p: 0
}
rectification_matrix: 1
rectification_matrix: 0
rectification_matrix: 0
rectification_matrix: 0
rectification_matrix: 1
rectification_matrix: 0
rectification_matrix: 0
rectification_matrix: 0
rectification_matrix: 1
```

The correct intrinsic matrix would look like this though:
```
[[351.05947319   0.         320.        ]
 [  0.         368.86247951 240.        ]
 [  0.           0.           1.        ]]
```

azeey, 2024-04-04:
I'm not able to reproduce this. After removing `vertical_fov`, since it's not in the [SDF spec](http://sdformat.org/spec?ver=1.11&elem=sensor#sensor_camera), with `gz-sim 8.2.0`, I get:

```

header {
  stamp {
    sec: 49
    nsec: 300000000
  }
  data {
    key: "frame_id"
    value: "camera_link_optical"
  }
}
width: 640
height: 480
distortion {
  k: 0
  k: 0
  k: 0
  k: 0
  k: 0
}
intrinsics {
  k: 351.05941772460938
  k: 0
  k: 320
  k: 0
  k: 351.05941772460938
  k: 240
  k: 0
  k: 0
  k: 1
}
projection {
  p: 351.05941772460938
  p: 0
  p: 320
  p: 0
  p: 0
  p: 351.05941772460938
  p: 240
  p: 0
  p: 0
  p: 0
  p: 1
  p: 0
}
rectification_matrix: 1
rectification_matrix: 0
rectification_matrix: 0
rectification_matrix: 0
rectification_matrix: 1
rectification_matrix: 0
rectification_matrix: 0
rectification_matrix: 0
rectification_matrix: 1

```

wittenator, 2024-04-04:
Oh wow, I don't know where I got this property from. Thanks for the quick fix! I assume that the vertical fov is computed via the horizontal fov and the image width and height given that the virtual sensor has quadratic pixels?

azeey, 2024-04-04:
I wasn't able to reproduce the issue even with `<vertical_fov>` in there. I just thought I would mention it.

>  I assume that the vertical fov is computed via the horizontal fov and the image width and height given that the virtual sensor has quadratic pixels?

I believe the vertical fov is determined by the horizontal fov and the aspect ratio (see https://gazebosim.org/api/rendering/8/classgz_1_1rendering_1_1Camera.html#adc6fc0b99e5c81f89408f0a69693f155), which is set to `width/height`: https://github.com/gazebosim/gz-sensors/blob/27239b4ac9877d87273a18010a307172899dd7a6/src/CameraSensor.cc#L257

I'll go ahead and close this issue since it doesn't appear to be a bug in gz-sensors

closed 2024-04-04 by azeey