gazebosim/gz-sensors #425: Magnetometer not taking into account the initial global heading

5p00kk, 2024-04-09:
The magnetometer sensor/system ignores the initial heading set from spherical coordinates. 

As seen in the lines below the world frame magnetic field vector is rotated from the world frame into local frame, however the spherical coordinates also hold information about the heading of the world coordinate system in regards to NED/ENU. The initial heading is not taken into account:

https://github.com/gazebosim/gz-sensors/blob/27239b4ac9877d87273a18010a307172899dd7a6/src/MagnetometerSensor.cc#L162C3-L166C1

This is incorrect and not consistent with the approach taken for the IMU sensor, where the system checks the initial heading and passes it to the sensor to take it into account. The issue makes it impossible to sync align and magnetometer sensors if the world coordinate heading is non-zero:
https://github.com/gazebosim/gz-sim/blob/c158287d6f8556112d86d6e4188c39784364bd40/src/systems/imu/Imu.cc#L228C2-L248C4

It seems like an issue that would require changes in both: gz-sim (system) and gz-sensor (sensor) side if following the approach taken for the IMU sensor.

azeey, 2024-07-15:
Thanks @5p00kk. I've added the "help wanted" tag on this. Let me know if you'd like to work on it.

azeey, 2025-09-10:
No updates, unfortunately. We'd appreciate any help on this.

Dylan-Gallagher, 2026-08-17:
I traced this against current `gz-sensors` / `gz-sim` and would like to work on it, but there is one compatibility choice worth confirming before I change the frame semantics.

`MagnetometerSensor` already treats `SetWorldMagneticField(...)` as a vector in the Gazebo world frame and applies the sensor-pose inverse rotation. The missing conversion appears to be at the `gz-sim` call site: SDF defines `<magnetic_field>` in the frame established by `<spherical_coordinates>`, while the system passes that vector through unchanged. For ENU input, `SphericalCoordinates::LocalFromGlobalVelocity(...)` provides the heading-adjusted Gazebo-world vector directly, so the fix can remain in `gz-sim` without adding a second frame transform to `gz-sensors`.

The wrinkle is the later `use_earth_frame_ned` compatibility option from gazebosim/gz-sim#2460. Its current default deliberately preserves NED-valued output even though Gazebo's world frame is ENU. A full ENU-to-local transform would make the standards-correct `false` path straightforward, but applying it to the legacy `true` path would also change that flag's zero-heading axis convention.

My proposed regression matrix is:

- explicit world `<magnetic_field>`, heading 0° / 90°;
- generated field, `use_earth_frame_ned=false`, heading 0° / 90°;
- sensor yaw separated from world heading so the two rotations cannot cancel;
- comparison with the IMU named-frame heading behavior.

Would you prefer that the fix:

1. preserves the legacy NED axis convention and applies only the heading yaw to that path, while using the full ENU-to-local conversion for `use_earth_frame_ned=false`; or
2. introduces an opt-in compatibility parameter and makes all corrected paths represent a true Gazebo-world vector?

Once that compatibility boundary is confirmed, I can implement the bounded `gz-sim` change and tests.