gazebosim/gz-sensors #454: I tested running fortress with the current `ign-sensors6` branch, and `<ignition_frame_id>` works for me. Do you have a minimal example that shows `<ignition_frame_id>` is not working?

MrKeith99, 2024-08-14:
## Summary
Related pull request: https://github.com/gazebosim/gz-sensors/pull/452

Related issue: https://github.com/gazebosim/gz-sensors/issues/239

> the frame id is `my_frame_id` which is what I set in the `<ignition_frame_id>` field in depth_frame_id.sdf

_Originally posted by @iche033 in https://github.com/gazebosim/gz-sensors/issues/452#issuecomment-2286831195_

Thank you for your prompt feedback.

This is the environment I am currently using:
- OS Version: Ubuntu 22.04
- Binary build
- version: Fortress version 6.16.0

I have tested your given sample code, achieving the following result:

```
$ ign topic -e -t /depth/points | grep header -A 8
header {
  stamp {
    sec: 7
    nsec: 783000000
  }
  data {
    key: "frame_id"
    value: "depth_camera::link::camera"
  }
```

But no problem with the `/depth` topic
```
$ ign topic -e -t /depth | grep header -A 8
header {
  stamp {
    sec: 743
    nsec: 347000000
  }
  data {
    key: "frame_id"
    value: "my_frame_id"
  }
```

This occurs in the same way with the robot that I am trying to simulate using xacro and linking it with gazebo using `reference` and `ignition_frame_id` tags.

MrKeith99, 2024-08-14:
In detail, this is my gazebo-related xacro file.
```xml

    <!-- Sensor: Depth -->
    <gazebo reference="head_camera_depth_optical_frame">
      <sensor name="head_camera_depth" type="depth">
        <ignition_frame_id>head_camera_depth_optical_frame</ignition_frame_id>
        <always_on>1</always_on>
        <update_rate>10</update_rate>
        <visualize>0</visualize>
        <topic>/head_camera/depth</topic>
        <pose>0 0 ${0.0175/2} ${M_PI/2} -${M_PI/2} 0 </pose>
        <camera name="d435">
          <horizontal_fov>1.48702</horizontal_fov>
          <image>
            <width>1280</width>
            <height>720</height>
          </image>
          <clip>
            <near>0.1</near>
            <far>10</far>
          </clip>
          <noise>
            <type>gaussian</type>
            <mean>0</mean>
            <stddev>0.1</stddev>
          </noise>
        </camera>
      </sensor>
    </gazebo>

    <!-- Ignition Sensors -->
    <gazebo>
      <plugin filename="libignition-gazebo-sensors-system.so" name="ignition::gazebo::systems::Sensors">
      </plugin>
    </gazebo>
```

And this is the tf tree
[image]

This are the results for the `/depth/points` data:
```
$ ign topic -e -t /head_camera/depth/points | grep header -A 8
header {
  stamp {
    sec: 5
    nsec: 200000000
  }
  data {
    key: "frame_id"
    value: "sobit_light::head_tilt_link::head_camera_depth"
  }
```

And for the `/depth` data:
```
$ ign topic -e -t /head_camera/depth | grep header -A 8
header {
  stamp {
    sec: 31
    nsec: 600000000
  }
  data {
    key: "frame_id"
    value: "head_camera_depth_optical_frame"
  }

```

iche033, 2024-08-15:
ok I was able to reproduce this with ignition-sensors installed from the debs, version 6.8.0, which was released in Jan 2024. I don't see the issue when building from source. I think the depth camera point cloud frame id fix is in: https://github.com/gazebosim/gz-sensors/pull/417. We should make a new sensors release for fortress

MrKeith99, 2024-08-16:
Great.
Thank you!
Ill try building from source.

DanielBar98, 2024-09-16:
@iche033 Do you have an approximate estimate of when the new release can be expected?

iche033, 2024-09-18:
we're currently focused on getting the upcoming Gazebo Ionic release out the door. Once that's out (end of Sept). I can make another ign-sensors6 release

iche033, 2024-10-04:
6.8.1 release is out (https://github.com/gazebosim/gz-sensors/pull/475)

closed 2024-10-04 by iche033

MrKeith99, 2024-10-05:
@iche033 
Thank you! I will test it asap!