gazebosim/gz-sensors #455: Slow simulation until cameras provide correct FPS

EmmanuelMess, 2024-08-15:
I am running a gazebo plugin for simulating the Intel Realsense D435i. When I record a rosbag, the rate of frames is ridicoulously low (3.5Hz). By reducing the `real_time_update_rate` to 10 I can get something similar to 30Hz (what it should be), but I want to set the simulation so that everything runs as slow as needed to keep the framerate at some specified value.

Plugin code: https://github.com/m-tartari/realsense_gazebo_plugin/tree/49ab134fa517280bb8393e715173dc631ccc2557

Similar to https://github.com/gazebosim/gz-sensors/issues/332

EmmanuelMess, 2024-08-15:
Workaround is to lower real_time_update_rate until the framerate reaches 30fps.

azeey, 2024-08-15:
Based on your use of `realsense_gazebo_plugin`, it sounds like you're using [Gazebo-classic](https://gazebosim.org/about). This repo is for the new Gazebo. Can you confirm?

EmmanuelMess, 2024-08-15:
> Based on your use of `realsense_gazebo_plugin`, it sounds like you're using [Gazebo-classic](https://gazebosim.org/about). This repo is for the new Gazebo. Can you confirm?

I'm using gazebo classic. Where is the repo for that version?

azeey, 2024-08-15:
https://github.com/gazebosim/gazebo-classic. I will mention that Gazebo-classic will be EOL in January, 2025 and there's minimal maintenance going on right now. I encourage you to switch to modern Gazebo (see https://gazebosim.org/docs/latest/migrating_gazebo_classic_ros2_packages/)

closed 2024-08-15 by azeey