gazebosim/gz-sensors #473: Issue visualizing Depth image in rviz2

EnzoGhisoni, 2024-09-27:
## Environment
* OS Version: Ubuntu 22.04
* Binary build
* Version: Fortress version 6.16.0 and ROS2 Humble

## Description
* Expected behavior: 
I try to add the depth sensors in the sdf of my robot in order to have the depth image available in ROS2.
* Actual behavior:
I have the good output for the depth image in Gazebo Image display viewer:
[image]

But when I visualize it in rviz2 the image is fully black or is in reverse color pattern in function of the settings:
[image]

[image]

## Steps to reproduce

I have implement the sensor in this way in my model.sdf:
`
<sensor name="camera_center" type="depth_camera">
      <gz_frame_id>camera_base_link</gz_frame_id>
      <always_on>1</always_on>
      <update_rate>30</update_rate>
      <visualize>0</visualize>
      <topic>/model/diff_drive/depth_camera</topic>
      <pose>0 0 0 0 -0 0</pose>
      <camera name="camera">
        <horizontal_fov>1</horizontal_fov>
        <image>
          <width>640</width>
          <height>360</height>
          <format>RGB_INT8</format>
        </image>
        <clip>
          <near>0.3</near>
          <far>50.0</far>
        </clip>
      </camera>
  </sensor>
`
I have this for the bridge:
`
- ros_topic_name: "/depth"
  gz_topic_name: "/model/diff_drive/depth_camera"
  ros_type_name: "sensor_msgs/msg/Image"
  gz_type_name: "gz.msgs.Image"
  direction: GZ_TO_ROS
`

I'm not sure if the difference come from the encoding, the bridge or simply the rviz2 configuration (if it is it could be interesting to know the default settings to visualize the data).

iche033, 2024-10-09:
I think it's just that the "visualization" is different but the depth image data encoding and depth values should be the same.  The image consists of float32 depth values. In gazebo, from [this description](https://github.com/gazebosim/gz-gui/blob/d110e5b288e8b580b28c0a7256d0e1f0e0c402e2/src/plugins/image_display/ImageDisplay.cc#L130), large depth values are darker (black) while smaller depth values are brighter (white), which just happens to be opposite of how rviz visualizes depth data.

azeey, 2024-12-16:
@EnzoGhisoni I'm going to assign the "help wanted" label to this. If you're able to submit a PR with the suggested changes in https://github.com/gazebosim/gz-sensors/issues/473#issuecomment-2457773426, we would greatly appreciate it.

exMachina316, 2025-03-30:
I would like to take up this issue.

closed 2025-04-04 by iche033