gazebosim/gz-sensors #488: Provide a way to set  the optical frame convention used for rendering sensors

azeey, 2024-12-11:
## Desired behavior
Currently, Gazebo uses the +x forward convention for all rendering sensors. This was done to keep the same behavior as Gazebo Classic. However, this is at odds with the convention in ROS, which is `+z` forward.  Users run into this issue (e.g. https://github.com/gazebosim/gz-sensors/issues/29) and the current workaround is to set a static transform in ROS to create a separate optical frame. There's an old PR in https://github.com/gazebosim/ros_gz/pull/257 to add a dedicated publisher, but I think it would be best if Gazebo itself provided a way to set the convention used. 

## Implementation suggestion
We could do one of the following:

1. Add an SDFormat tag to all rendering sensors that sets whether the `+z` convention should be used.
2. Add an SDFormat under `<sensor>` that sets whether the `+z` convention should be used. This would be easier since we'd only modify one SDFormat description file, but it would affect all sensors
3. One of the two options above, but the tag would allow setting a custom transformation matrix.

Sukhvansh2004, 2025-01-19:
Hi, can I work on this issue?

azeey, 2025-01-21:
@Sukhvansh2004, thanks for volunteering. I'll assign the ticket to you. @iche033, @mjcarroll any thoughts on which option we should follow?

iche033, 2025-01-21:
I wrote some notes down not long ago on what sensors need optical frames:

>  Regarding optical frames, I looked back at what we did in subt and mbzirc and this is what I think:
> * we only need to use optical frame for regular RGB camera and depth camera images, e.g. /camera/image_raw and /camera/depth  topics. As an example, this was done in mbzirc: https://github.com/osrf/mbzirc/blob/main/mbzirc_ign/src/mbzirc_ign/model.py#L235-L260. We only created optical frames for image_raw and depth topics.
> * we don't need optical frames for point cloud topics, i.e. lidar point clouds and RGBD camera's point clouds, e.g. /lidar/points  or /rgbd_camera/points topics

The `optical frame publisher` sets this [transformation pose](https://github.com/gazebosim/ros_gz/pull/257/files#diff-f8a9d04c891ede310d8b02453045ded0a0bcb3b9f8d404c7799dd6555e987f86R163-R176) to create a z forward optical frame.

So if the above is true, instead of updating the generated sensor data itself, I think we just need a way to add an optical link with the above transformation inside gz-sim and set the msg frame id to that optical link name. 

Here's a proposal that's similar to option 2.

```xml
<sensor>
  <!-- Use the existing <frame_id> element (sdf 1.12+) to specify frame as usual -->
  <frame_id>my_z_optical_frame</frame_id>
  <!-- Support a new //sensor/frame element.
       Gazebo sees this and creates a new frame (link?) with the specified pose -->
  <frame name="my_z_optical_frame">
    <pose>0 0 0 -1.5708 0 -1.5708</pose>
  </frame>
</sensor>
```

Open to other ideas.

JorgePH, 2025-01-22:
I think the issue in #29 was that the _frame_id_ in the RGBD pointcloud was populating with the _optical_frame_id_ too (there is only one field to define the frame in RGBD sensor). As long as the pcd (following ROS convention) and cameras (following cameras convention) don't get published on different frames, their information is going to be conflicting. Thanks to ROS tf tools, as long as you have the tf tree well defined, it does not matter too much the frame where the data is published as long as it is consistent.

IMO at least another frame tag should be added for RGBD.

Danilrivero, 2025-01-22:
> I think the issue in [#29](https://github.com/gazebosim/gz-sensors/issues/29) was that the _frame_id_ in the RGBD pointcloud was populating with the _optical_frame_id_ too (there is only one field [...]

I agree on this. Your camera [(e.g)](https://github.com/IntelRealSense/realsense-ros/tree/ros2-master#ros2robot-vs-opticalcamera-coordination-systems) should provide you with the required transformations following the proper conventions and use this information to properly transform to any frame required (which would be on your side).

In this case Gazebo would be expecting this format and another frame would be required to fully define your RGBD camera.

Open to discuss this and see if it could be a possibility.

iche033, 2025-01-22:
Based on the comments above and in #29, The issue came from https://github.com/gazebosim/gz-sensors/pull/458 (and https://github.com/gazebosim/gz-sensors/pull/362 for DepthCameraSensor) since they incorrectly set the point cloud msg to optical frame. These PRs were supposedly submitted to fix "optical frame issues" for their use case. However, after gathering these recent comments I agree we should provide the ability to use a separate frame for the point cloud msg. 

One example would be to use `<frame_id>` and `<optical_frame_id>` for the separate frames:

```xml
<!-- sdf 1.12 -->
<sensor name="my_sensor" type="rgbd">
  <!-- point clouds will use this frame -->
  <frame_id>sensor_frame</frame_id>
  <camera>
    <!-- images will use this frame -->
    <optical_frame_id>optical_frame<opical_frame_id>
    ...
  </camera>
...
</sensor>
```

To achieve the above, we'll unfortunately need to revert #458 and #362, and un-deprecate the `<optical_frame_id>` tag in sdf 1.12. Not ideal since we're changing behavior yet again so I want to be very sure this is the right fix. Open to suggestions.

Sukhvansh2004, 2025-01-22:
> I agree on this. Your camera [(e.g)](https://github.com/IntelRealSense/realsense-ros/tree/ros2-master#ros2robot-vs-opticalcamera-coordination-systems) should provide you with the required [...]

Yeah I myself have worked on depth camera once and had to figure out that the external sensor frame tag needs to be as per the XYZ convention while the internal plugin frame needs to follow the ROS convention of ZYX

Sukhvansh2004, 2025-01-22:
I have some doubts too regarding the implementations, currently I am a bit occuped with some work will try to get in contact and start working on this issue ASAP this weekend

iche033, 2025-01-23:
Thanks for helping to work on this. Before going straight into implementation, I would like to get some feedback on the proposal first. I talked to @azeey about this and I think he has other thoughts on how we may want to address this from `ros_gz`.

Danilrivero, 2025-01-23:
> Based on the comments above and in [#29](https://github.com/gazebosim/gz-sensors/issues/29), the issue came from [#458](https://github.com/gazebosim/gz-sensors/pull/458) (and [...]

> To achieve the above, we'll unfortunately need to revert [#458](https://github.com/gazebosim/gz-sensors/pull/458) and [#362](https://github.com/gazebosim/gz-sensors/pull/362), and un-deprecate the [...]

I agree with you, even though the revert could cause some trouble it seems like the path we would go with, having the `<optical_frame_id>` tag implemented. Some sort of backwards compatibility could be discussed if required for any possible breaking changes.

> Thanks for helping to work on this. Before going straight into implementation, I would like to get some feedback on the proposal first. I talked to [@azeey](https://github.com/azeey) about this and I think he has other thoughts on how we may want to address this from `ros_gz`.

Any information on this?

Danilrivero, 2025-02-06:
Hi, I’m following up on this issue. Are there any updates or additional information @iche033?

iche033, 2025-02-06:
> Hi, I’m following up on this issue. Are there any updates or additional information [@iche033[[image]](https://teams.googleplex.com/iche@google.com)](https://github.com/iche033)?

@azeey, any thoughts?

azeey, 2025-02-10:
My main issue with using `<optical_frame_id>` is that it makes `frame_id` almost unnecessary. By that I mean, every `Image` and `CameraInfo` message would use the `optical_frame_id` instead of `frame_id` since that's what would be required in ROS. The only message that would use `frame_id` would be `PointCloud`. As we've seen in the past, this causes confusion because the most logical tag to specify the frame (`frame_id`) is almost always ignored. Also, since the frame specified in `optical_frame_id` would not actually exist in Gazebo as a frame (usually this would only exist in ROS via a static transform), consuming the `Image` message from other Gazebo systems/plugins would be problematic.

Generally, I think we should keep using the `frame_id` tag and the convention that we've used in Gazebo for frames. When bridging to ROS, that's when we should create the optical frame and convert to the ROS frame conventions. Ideally, the creation of an optical frame would be handled on the `ros_gz` bridge since it's only needed when using Gazebo with ROS. From the Gazebo side, we'd only have the `frame_id` tag and `ros_gz` would attach a `_optical` to the string used in `frame_id` and publish the `Image` and `CameraInfo` topics. It would also publish the necessary static transform to `/tf`.

peci1, 2025-02-11:
I think the proposal to solve it in ros_gz is going in a good direction. However, I'm not sure if at the ros_gz level, there is enough information to correctly process it.

I'll try to explain.

Each (physical) camera has (in my view) 4 frames (the terminology is mine, do not try to google it):

1. The *mechanical* frame. This is a XYZ frame defining how the camera is mounted to the robot. Usually, this frame is in the middle of some mounting surface, in a tripod screw or somewhere similar. In Gazebo, it is usually a `<link>` with geometry and collisions.
2. The *geometrical* frame. This is a XYZ frame that denotes the optical center of the pinhole lens. I've seen implementations where this is an empty `<link>` in Gazebo and other implementations where this is `<sensor><pose>`.
3. The *optical* frame. This is a ZYX frame where the image lies in plane Z=0 and has the standard image coordinates Y down X right. This is what is proposed to be set by the `<optical_frame_id>` or its successor.
4. The *optical XYZ* frame. This is a XYZ frame that has the 90-0-90 rotation from optical frame.

You might be confused by the 4 frames and not 3. The reason why there are 4 frames is that e.g. in case the camera is mounted upside-down, the camera driver can be set to rotate the image by 180 degrees. Many cameras support this and many real use-cases do it so that image processing algorithms do not need to rotate the image. This rotation goes exactly between *geometrical* and *optical XYZ* frames. In the simple case with no rotation, both *geometrical* and *optical XYZ* frames are identical and that's why most people only think about 3 frames.

RGBD or multi-lens sensors would have a single *mechanical frame* and the other frames would be one for each lens.

In my ideal world, the assignment of the transforms in Gazebo would be:

1. *mechanical* = TF from `link/pose`, name from `link/name`
2. *geometrical* = TF from `sensor/pose`, name from `sensor/frame_id`
3. *optical XYZ* = TF from `camera/pose`, no name
4. *optical* = TF from `camera/pose` * 90-0-90, name from `camera/optical_frame_id` (could default to `sensor/frame_id + _optical_frame`)

This would be used for all pinhole sensors like RGB camera, Thermal camera and depth camera images (). Sensors like lidar and depth camera pointcloud would be happy with just *mechanical* and *geometrical* frames (thus utilizing `sensor/frame_id`). RGBD sensors could use the *optical* frame for images and *geometrical* frame for pointclouds.

Now, the problem is Gazebo would need the images published with *optical XYZ* frame (for which I don't have a name), while ROS would need them with *optical* frame. However, as the header in Gazebo messages is a KV store, we could easily store both (or all three) frames there.

(Un)fortunately, this touches a little bigger problem. That is transferring named frames from Gazebo to ROS. If something hasn't changed, only joints and model poses are transferred to ROS TFs (and only if you set up a TF bridge which is unfortunately undocumented anywhere (after my quick search for "ros_gz tf bridge")). However, as Gazebo also got the option of defining named frames, it would be useful to get these transferred to ROS, too.

If such mechanism were available, it seems to me a solution to this problem would be almost easy. `sensor/frame_id` would generate a named frame, `camera/optical_frame_id` would also generate a named frame and these frames would be published to ROS. This would alleviate users from having to either set up static TF publishers in ROS or code some C++ plugins for optical frame publishers. Of course it has problems with things like frame namespacing etc., but I guess the general idea should be clear, is it?

As a bonus, this would render unnecessary the current approaches I've seen where the optical center of the camera is represented by an empty link, just to get this frame to ROS. However, empty links connected via fixed joints make the whole simulation unstable. I've seen for example a camera sensor flying around a 2-ton robot I was designing. The only way to work it around was specifying the camera an unreasonable weight like 1 kg, where it has 40 g in reality. If we could transfer named frames to ROS, many of these empty links would just disappear.

Danilrivero, 2025-02-12:
> My main issue with using `<optical_frame_id>` is that it makes `frame_id` almost unnecessary. By that I mean, every `Image` and `CameraInfo` message would use the `optical_frame_id` instead of [...]

I can agree that this issue is due to the optical frame convention in ROS, which is not the whole scope of Gazebo.

> Generally, I think we should keep using the `frame_id` tag and the convention that we've used in Gazebo for frames. When bridging to ROS, that's when we should create the optical frame and convert [...]

This seems like a proper way to deal with it for ROS conventions, so it's an OK from me.

iche033, 2025-02-15:
> However, as Gazebo also got the option of defining named frames, it would be useful to get these transferred to ROS, too.

In past projects we used the [pose_publisher](https://github.com/gazebosim/gz-sim/tree/gz-sim9/src/systems/pose_publisher) system  and the ros_gz bridge to convert poses of gz entities to ROS 2 TFs. One option for transferring frames to ROS is to extend the `pose_publisher` system to support publishing pose of [frames defined in models](http://sdformat.org/spec?ver=1.12&elem=model#model_frame), with a configurable `<publish_frame_pose>` plugin param that defaults to `false`

dcconner, 2025-02-18:
Just stumbled onto this.  Is there a reason that the points are not published in the same frame as the image and depth_image on Gazebo Harmonic ?   I was fighting this for a bit trying to convert some old gazebo classic files, and finally found the optical_frame_id that fixed my issues with image and image_depth, then I just discovered that the points are not aligned with the depth_image.  Is there a use case for the points message to be in a different frame from the depth map.

In this image the laser scan and depth_image are correct, while point cloud is incorrect, even though they all report the designated `camera_depth_optical_frame`

[image]

iche033, 2025-03-13:
created a draft PR in https://github.com/gazebosim/ros_gz/pull/703 for the solution proposed in https://github.com/gazebosim/gz-sensors/issues/488#issuecomment-2649312331