gazebosim/gz-sensors #489: [Err] [WideAngleCameraSensor.cc:408] WideAngleCamera doesn't exist.

drfenixion, 2024-12-22:
## Environment
* OS Version: Ubuntu 22.04.5 LTS
* Source or binary build?

As binary:
Gazebo Garden
 
  As source:
[submodule "modules/ros_gz"]
	path = modules/ros_gz
	url = https://github.com/gazebosim/ros_gz
	branch = iron
[submodule "modules/gz_ros2_control"]
	path = modules/gz_ros2_control
	url = https://github.com/ros-controls/gz_ros2_control
	branch = iron

## Description
* Expected behavior: Select topic of wideanglecamera in ImageDisplay widget of Gazebo (other camera types work this way) and want to see video from camera
* Actual behavior:  got gazebo errors ([Err] [WideAngleCameraSensor.cc:408] WideAngleCamera doesn't exist.)

## Steps to reproduce
I use wide angle camera SDF example from official exmaples

        <sensor name="wideangle_camera" type="wideanglecamera">
          <camera>
            <horizontal_fov>3.14</horizontal_fov>
            <image>
              <width>800</width>
              <height>600</height>
            </image>
            <clip>
              <near>0.1</near>
              <far>100</far>
            </clip>
            
            <lens>
              
              <type>custom</type>
              
              
              <custom_function>
                <c1>1.05</c1>
                
                <c2>4</c2>
                
                <f>1.0</f>
                
                <fun>tan</fun>
                
              </custom_function>
              
              
              <scale_to_hfov>true</scale_to_hfov>
              
              <cutoff_angle>3.1415</cutoff_angle>
              
              <env_texture_size>512</env_texture_size>
            </lens>
          </camera>
          <always_on>true</always_on>
          <update_rate>30</update_rate>
          <topic>wide_angle_camera</topic>
        </sensor>
        
       https://github.com/gazebosim/gz-sim/blob/gz-sim7/examples/worlds/wide_angle_camera.sdf
       i also tried other sdf configuration and they gave same error.
       
       I also  tried some other camera types they work correct.

## Output
[ruby $(which gz) sim-4] [Err] [WideAngleCameraSensor.cc:408] WideAngleCamera doesn't exist.
[image]

azeey, 2025-01-07:
I tested using gz-sim 7.9.0 and I couldn't reproduce this problem. What version are you running? You can find out using `gz sim --version`.

I just ran `gz sim -v4 wide_angle_camera.sdf`.

[image]

drfenixion, 2025-01-16:
Gazebo 7.9.0

I changed some SDF parameters  and now it returns meaningfull error (looks like with some options it can returns silent error like "WideAngleCamera doesn't exist."). Reason is Ogre2 does not support Wide angle camera. 
```
[ruby $(which gz) sim-4] [Err] [BaseScene.hh:704] Wide angle camera not supported by: ogre2
[ruby $(which gz) sim-4] [Err] [WideAngleCameraSensor.cc:231] Failed to create wide angle camera
[ruby $(which gz) sim-4] [Err] [SceneManager.cc:1763] Unable to find sensor 
[ruby $(which gz) sim-4] [Err] [RenderUtil.cc:1318] Failed to create sensor 
```

On Ogre (1) it works correct.

iche033, 2025-01-24:
wide angle camera support was added in gz-rendering8 (harmonic): https://github.com/gazebosim/gz-rendering/pull/733