gazebosim/gz-sensors #503: 🧑‍🌾

Crola1702, 2025-02-18:
## Environment
* OS Version: Windows conda legacy
* Source gz-sensors6 build

## Output

Reference build:
* https://build.osrfoundation.org/job/gz_sensors-ign-sensors6-clwin/6

Test regressions:
* [DepthCameraSensor/DepthCameraSensorTest.ImagesWithBuiltinSDF/ogre2](https://build.osrfoundation.org/job/gz_sensors-ign-sensors6-clwin/6/testReport/junit/(root)/DepthCameraSensor_DepthCameraSensorTest/ImagesWithBuiltinSDF_ogre2/)
* [RgbdCameraSensor/RgbdCameraSensorTest.ImagesWithBuiltinSDF/ogre2](https://build.osrfoundation.org/job/gz_sensors-ign-sensors6-clwin/6/testReport/junit/(root)/RgbdCameraSensor_RgbdCameraSensorTest/ImagesWithBuiltinSDF_ogre2/)
* [ThermalCameraSensor/ThermalCameraSensorTest.ImagesWithBuiltinSDF/ogre2](https://build.osrfoundation.org/job/gz_sensors-ign-sensors6-clwin/6/testReport/junit/(root)/ThermalCameraSensor_ThermalCameraSensorTest/ImagesWithBuiltinSDF_ogre2/)
* [ThermalCameraSensor/ThermalCameraSensorTest.Images8BitWithBuiltinSDF/ogre2](https://build.osrfoundation.org/job/gz_sensors-ign-sensors6-clwin/6/testReport/junit/(root)/ThermalCameraSensor_ThermalCameraSensorTest/Images8BitWithBuiltinSDF_ogre2/)

<details>
<summary>
depth_camera
</summary>

```
C:\J\workspace\gz_sensors-ign-sensors6-clwin\ws\gz-sensors\test\integration\depth_camera.cc:369
Expected equality of these values:
  g_depthBuffer[mid]
    Which is: inf
  -gz::math::INF_D
    Which is: -inf
```

</details>

<details>
<summary>
rgbd_camera
</summary>

```
C:\J\workspace\gz_sensors-ign-sensors6-clwin\ws\gz-sensors\test\integration\rgbd_camera.cc:583
Expected equality of these values:
  -math::INF_D
    Which is: -inf
  g_depthBuffer[index]
    Which is: inf
```

</details>

<details>
<summary>
thermal_camera
</summary>

```
C:\J\workspace\gz_sensors-ign-sensors6-clwin\ws\gz-sensors\test\integration\thermal_camera.cc:312
The difference between boxTemp and g_thermalBuffer[mid] * linearResolution is 14, which exceeds boxTempRange, where
boxTemp evaluates to 310,
g_thermalBuffer[mid] * linearResolution evaluates to 296, and
boxTempRange evaluates to 3.

...

C:\J\workspace\gz_sensors-ign-sensors6-clwin\ws\gz-sensors\test\integration\thermal_camera.cc:574
The difference between boxTemp and g_thermalBuffer8Bit[mid] * linearResolution is 13, which exceeds linearResolution, where
boxTemp evaluates to 310,
g_thermalBuffer8Bit[mid] * linearResolution evaluates to 297, and
linearResolution evaluates to 3.
```

</details>

Crola1702, 2025-04-21:
Friendly ping @iche033

iche033, 2025-05-23:
given that it's fortress on windows, I may just disable those tests