gazebosim/gz-sensors #509: Far lasers appear to hit the floor

atinfinity, 2025-03-23:
## Environment

* OS Version: Ubuntu 24.04
* Source or binary build?: binary build, Gazebo Harmonic v8.9.0
  
  

## Description

[This URDF](https://github.com/ros-navigation/nav2_minimal_turtlebot_simulation/blob/jazzy/nav2_minimal_tb4_description/urdf/sensors/rplidar.urdf.xacro) has the following description.

```xml
<xacro:ray_sensor sensor_name="${name}" gazebo="${gazebo}" 
                update_rate="10.0" visualize="true" 
                h_samples="360" h_res="1.0" h_min_angle="0.000" h_max_angle="6.280000" 
                r_min="0.164" r_max="20.0" r_res="0.01">
                <plugin name="dummy" filename="dummyfile"></plugin>
</xacro:ray_sensor>
```

The 2D LiDAR is mounted horizontally to the ground.

* Expected behavior:
  * Lasers does not hit the floor
* Actual behavior:
  * Lasers hit the floor

## Steps to reproduce

I installed ROS 2 Jazzy and nav2_bringup packge.

```bash
sudo apt install ros-jazzy-desktop-full ros-jazzy-nav2-bringup
```

I launched `tb3_simulation_launch.py` using large world(`depot.sdf`).

```bash
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False autostart:=False world:=/opt/ros/jazzy/share/nav2_minimal_tb4_sim/worlds/depot.sdf
```

## Output

[image]

pranavk-2003, 2025-04-09:
Managed to identify and solve the issue, this issue need to be in [Nav2](https://github.com/ros-navigation/navigation2) not in gz-sensors.

atinfinity, 2025-04-10:
Thank you! I closed this ticket.

closed 2025-04-10 by atinfinity