gazebosim/gz-sensors #517: setting imu frame

PerFrivik, 2025-04-18:
Hey I'm using gz harmonic, and im really struggeling to set the frame id... it always shows me the full path: 

header {
  stamp {
    sec: 5209
    nsec: 518000000
  }
  data {
    key: "frame_id"
    value: "smb::imu_link::imu_sensor"
  }
  data {
    key: "seq"
    value: "2550031"
  }
}
entity_name: "smb::imu_link::imu_sensor"
orientation {
  x: -1.490030155022471e-12
  y: -5.2243644810182165e-13
  z: 0.943628091709525
  w: -0.331007589847484
}

But I just want it to be imu: 

    <gazebo reference="${imu_link}">
      <sensor name="imu" type="imu">
        <gz_frame_id>imu_sensor</gz_frame_id>
        
        
        <always_on>true</always_on>
        <update_rate>${imu_update_rate}</update_rate>
        <visualize>true</visualize>
        <topic>/imu</topic>
      </sensor>
    </gazebo>

what am I doing wrong? This somehow works for the lidar, just setting the gz_frame_id...

bentwookie, 2025-07-01:
I'd love to know as well.

AlexGisi, 2025-08-25:
Please see: https://github.com/gazebosim/gz-sensors/issues/306

I believe this issue should be closed